cleanflight/cleanflight #2254: Frsky SPort Telemetry not working when GPS enabled

renes, 2016-06-24:
I have the an X8R and a Taranis. I have smart port telemtry on software serial 5+6 which is working fine. The taranis will find all kind of sensors and gets updates.

but as soon as I enable gps on the 2nd soft serial the telemtry does not working any more. no updates anymore. GPS is working but telemetry not.

My configurations:

[cleanflight_backup_6.24.2016.11.46_telemtry_not_working.txt](https://github.com/cleanflight/cleanflight/files/331559/cleanflight_backup_6.24.2016.11.46_telemtry_not_working.txt)
[cleanflight_backup_6.24.2016.11.48_telemtry__working.txt](https://github.com/cleanflight/cleanflight/files/331558/cleanflight_backup_6.24.2016.11.48_telemtry__working.txt)

andre92, 2016-06-27:
I see your gps baud is 9600. Can you configure your gps baud at 57600? I think softserial makes both ports run at the same baud rate and smart port only works at 57600. I have a gps on softserial at 57600 and an X8R with smart port on softserial (connected to both rx and tx, and telemetry_inversion enabled) and both are working for me.

JohnOCFII, 2016-06-27:
> I see your gps baud is 9600. Can you configure your gps to run at 57600 baud? According to cleanflight documentation, all softserial ports will run at the same baud rate and smart port only works at 57600. I have a gps on softserial at 57600 and an X8R with smart port on softserial (connected to both rx and tx, and telemetry_inversion enabled) and both are working for me.

Interesting.  I didn't think SoftSerial was supported at speeds about 19200 -- something about being too processor intensive?  Perhaps on an F3 flight controller it is less of an issue (which would be great, btw)?

closed 2017-02-25 by hydra