cleanflight/cleanflight #2255: Weird ESC/motor problem

Prowlr, 2016-06-24:
I've been experiencing this problem that I haven't been able to figure out for weeks now. It's my first build and I haven't got it to a flyable point yet. Everything works apart from its ability to actually 

Once it was built I had my maiden flight. I arm, the motor spin at idle, i increase the throttle and the motors increase in speed by themselves and it flips aggressively. So I take the props off and test how the motors behave with different stick positions.

When I arm, the motors spin at idle like they should but when I adjust the throttle alone, it spins up and down like you would expect it to with all the motors in sync, but as soon as I put a little stick input that isn't throttle (yaw, roll or pitch) the motors begin to slowly increase until some of the motors are at max throttle and the others are about mid throttle. 

In CleanFlight, the motor bars go out of whack and the motors become out of sync. It's weird because when I switch flight modes in Cleanflight from horizon to angle to rate mode, the rate at which the throttle increases by itself changes. In rate mode, it goes up much slower than in horizon mode. I had the same issue with my KISS FC and switched to the Naze32 to try and figure out what it was. 

I've flashed and calibrated the ESCs through and I've searched online and couldn't find anything. 

All the connections are correct. Board orientation is correct. Can't see any shorts. Replaced the ESC that was causing me problems earlier but it made no difference.

I feel like it might be the PIDs but I have no idea how to tune them  
[Here is a video that shows what I see.](https://www.youtube.com/watch?v=5D3uHKBQbG8&feature=youtu.be)

Does anyone have any ideas?

andre92, 2016-06-24:
If you are worried about PIDs being too high, you can try lowering them to half the defaults for roll, pitch and yaw (P, I and D) the defaults are pretty high for small quads running BlHeli

Did you place the board with the arrow facing to the front of the quad? If not, it is easy to confuse. You have to set the correct orientation or all corrections will be applied in the wrong direction. (which causes agressive flips and 100% loss of control)

If you want to check board orientation, hold the copter in your hand, rate mode and NO props, and move it around. Basically any motor that is moving down should spin fast, so if you pitch forward (nose down), the front motors should spin up. Don't ever do this with props on, the copter could flip and bite you.

What I see in the video is normal, except your transmitter trims may be off center (they should be centered in the Receiver tab of the configurator) If it shows any jitter, set a small value for rc deadband.

Also, did you calibrate the accelerometer and verify the copter's orientation in the GUI while moving it around? When powering up or rebooting, hold the copter still for 20 seconds so it can get its initial orientation and calibrate the gyroscope.

Hope this helps, please let me know

Prowlr, 2016-06-25:
Thank you for your input. I double checked all those things and they were fine.

Turns out, the quad was working all along. I never bothered to actually fly it after I saw it flip on my maiden flight. Then once I saw the motors out of sync in CleanFlight I attributed the two together so I assumed I'd get the same results until I fix the out of sync motors. But nonetheless, it's working! I've waited literally months for this moment. Thanks again :)

closed 2017-02-25 by hydra