cleanflight/cleanflight #2257: Not arming when telemetry enabled on softserial

daJuels, 2016-06-25:
When I enable telemetry on Sofserial then arming is not possible..
I tried with connecting X4R SmartPort to Tx, to Tx&Rx, w/o Tx&Rx.

Please let me know if you need further information..

# Cleanflight Log
Running - OS: Windows, Chrome: 49.0.2623.112, Configurator: 1.2.2
Serial port successfully opened with ID: 1
MultiWii API version received - 1.19.0
Flight controller info, identifier: CLFL, version: 1.13.0
Running firmware released on: Jun 6 2016 00:03:48
Board: AFNA, version: 2

# dump

## version
## Cleanflight/NAZE 1.13.0 Jun  6 2016 / 00:03:48 (a77bc76)
## dump master

## mixer
mixer QUADX
mmix reset
smix reset

## feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature RX_PPM
feature VBAT
feature SOFTSERIAL
feature FAILSAFE
feature TELEMETRY
feature ONESHOT125

## map
map AETR1234

## serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 30 0 115200 57600 0 115200
serial 31 32 115200 57600 9600 115200

## led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
[... 400 lines cut ...]
1466864929394,1502,1499,1500,988,1996,1500,1995,1996,1500,1500,1500,1500,1000,3,0,0
1466864929894,1502,1499,1501,988,1996,1500,1995,1996,1500,1500,1500,1500,994,-3,0,0
1466864930394,1503,1500,1500,988,1995,1499,1996,1996,1500,1500,1500,1500,990,-8,0,0
1466864930894,1503,1499,1500,988,1996,1499,1996,1996,1500,1500,1500,1500,996,-1,0,0
1466864931394,1503,1500,1500,988,1995,1500,1996,1996,1500,1500,1500,1500,1002,5,0,0
1466864931894,1503,1499,1506,988,1995,1500,1995,1996,1500,1500,1500,1500,1000,3,0,0
1466864932394,1502,1499,1505,1053,1996,1500,1996,1996,1500,1500,1500,1500,997,0,0,0
1466864932894,1503,1499,1505,1052,1995,1500,1995,1995,1500,1500,1500,1500,994,-3,0,0
1466864933394,1503,1499,1505,1053,1996,1500,1996,1996,1500,1500,1500,1500,994,-3,0,0
1466864933894,1503,1499,1505,1053,1996,1996,1996,1996,1500,1500,1500,1500,1004,7,0,0
1466864934394,1503,1499,1505,1053,1996,1996,1996,1996,1500,1500,1500,1500,995,-2,0,0
1466864934894,1503,1500,1505,1052,1996,1995,1996,1996,1500,1500,1500,1500,993,-6,0,0
1466864935394,1503,1499,1505,1052,1995,1996,1996,1996,1500,1500,1500,1500,993,-5,0,0
1466864935894,1503,1500,1505,1052,1996,1500,1996,1996,1500,1500,1500,1500,1008,10,0,0
1466864936394,1503,1499,1651,2011,1996,1500,1996,1995,1500,1500,1500,1500,998,1,0,0
1466864936894,1453,1060,1579,988,1995,1500,1995,1996,1500,1500,1500,1500,990,-7,0,0
1466864937394,2011,1631,998,1883,1996,1500,1995,1996,1500,1500,1500,1500,992,-6,0,0
1466864937914,1065,993,2008,988,1996,1500,1995,1996,1500,1500,1500,1500,998,0,0,0
1466864938394,2006,1887,1385,2012,1996,1500,1995,1996,1500,1500,1500,1500,1003,6,0,0
1466864938894,1503,1498,1502,1086,1996,1500,1996,1996,1500,1500,1500,1500,994,-3,0,0
1466864939394,1503,1498,1501,988,1996,1500,1996,1996,1500,1500,1500,1500,1002,5,0,0
1466864939894,1503,1498,1501,988,1996,1500,1995,1996,1500,1500,1500,1500,999,2,0,0
1466864940394,1503,1499,1501,988,1995,1500,1995,1996,1500,1500,1500,1500,1003,5,0,0
1466864940894,1503,1498,1501,988,1996,1500,1995,1996,1500,1500,1500,1500,995,-2,0,0
1466864941394,1503,1498,1501,988,1996,1500,1996,1996,1500,1500,1500,1500,994,-3,0,0

Ede2016, 2016-06-26:
I can confirm this problem.
FC: AfroMini (MiniNaze32)
When I used CleanFlight 12.x I could use Telemetry on SoftSerial with SBus and arming worked in the way it should work.
When I try to do the same with CleanFlight 13.0 I can't arm my quad.
I did some more troubleshooting:
a) If I disable Telemetry arming works, SoftSerial is enabled with SBus but because Telemetry is off it is not really used
b) Disabling SoftSerial  does not make a difference. Just Telemetry ON, even if SoftSerial is OFF makes it impossible to arm
c) Arming with stick commands or with switch does not make a difference
d) Changing the loop time (I tried 1000 up to 4000) does not make a difference
e) Telemetry with SoftSerial works, I see all the values in my Taranis, I just can't arm the quad
f) Looking at the Modes Section in CleanFlight Configurator: I set a range for AUX2 to arm the quad. I can see in CleanFlight Configurator in the Modes section that the received signal is in the correct range but the ARM stays grey. If I switch off telemetry ARM is green when I use the correct switch position.
g) I tried other things to check exactly when the problem happens. Arming is not possible with Telemetry switched on. Nothing else makes a difference.

andre92, 2016-06-27:
Try cli commands 'tasks' and 'status' with said features enabled.
Maybe your cpu load is too high to arm.
It is a safety feature that should prevent bad things to happen in the air.
If your cpu load is too high, set gyro_sync_denom to 2, that will set looptime to 2000 and will fix cpu load

Ede2016, 2016-06-29:
@andre92 thanks for your info.
First: I set gyro_sync_denom = 2 and now it works.

I read about gyro_sync but somehow I did not realize that this automatically disables the looptime. In the Configurator it is still possible to set the looptime - the field is not disabled. I guess this is what confused me. I tried the 4000 just as a sample to have a very high value to verify if this makes a difference. But being disabled it obviously does not make a difference ;)

I had a look at the commands status and task list and I am a little confused. The task list shows me that baro uses about 36% and telemetry uses about 1%. With default gyro_sync_denom = 1 I disabled the baro and it does not show in the task list anymore but I still can't arm the quad. But if I disable telemetry which uses only 1% then I can arm it. Strange!

Last question: I tried to use the CLI commands status and tasks after I successfully armed the quad but it did not work. Is this normal? Is there any way to see the status and task list of the armed quad?

andre92, 2016-06-29:
Correct, telemetry itself uses only 1%. I believe that Softserial is responsible for a lot of cpu load since it uses so many interrupts, especially at high baud rates.
As for the baro, do you have a BMP280? Even though it shows a high percentage in tasks, I do not notice a difference in system load when I disable it. 
You are correct at CLI, it doesn't work when armed, but all tasks (except MSP) keep running in the background, so the load should not differ between armed and disarmed.

Ede2016, 2016-06-29:
Thanks @andre92 
Here is the info from my status and tasks:
This is with SoftSerial and Telemetry enabled, can't arm the quad (I understand why)

```
# status
System Uptime: 222 seconds, Voltage: 83 * 0.1V (2S battery - OK), System load: 1.87
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP280
Cycle Time: 991, I2C Errors: 0, registry size: 46

# tasks
Task list          max/us  avg/us rate/hz maxload avgload     total/ms
 0 -       SYSTEM      10       0       9    0.5%    0.5%        11
 1 -     GYRO/PID     934     732     897   84.2%   66.1%    169502
 2 -        ACCEL     178     144     450    8.5%    6.9%     17203
 3 -       SERIAL   44951       2      84  378.0%    0.5%       458
 4 -       BEEPER      15       0      62    0.5%    0.5%        31
 5 -      BATTERY   40099       1      43  172.9%    0.5%       147
 6 -           RX     281     171     111    3.6%    2.3%      4637
 9 -         BARO     143      75    4901   70.5%   37.2%      1061
11 -     ALTITUDE     350     177      38    1.8%    1.1%      1006
13 -    TELEMETRY      53       4     208    1.6%    0.5%       405
```

And here are the details after I set gyro_sync_denom = 2
```
# status
System Uptime: 18 seconds, Voltage: 81 * 0.1V (2S battery - OK), System load: 0.15
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP280
Cycle Time: 2005, I2C Errors: 0, registry size: 46

# tasks
Task list          max/us  avg/us rate/hz maxload avgload     total/ms
 0 -       SYSTEM      11       0       9    0.5%    0.5%         1
 1 -     GYRO/PID     899     726     499   45.3%   36.7%      7374
 2 -        ACCEL     181     144     847   15.8%   12.6%      1981
 3 -       SERIAL     754       2      99    7.9%    0.5%        85
 4 -       BEEPER      13       0      99    0.6%    0.5%         3
 5 -      BATTERY   40089       1      49  196.9%    0.5%        49
 6 -           RX     227     167     109    2.9%    2.3%       369
 9 -         BARO     151      75    6802  103.2%   51.5%       143
11 -     ALTITUDE     358     180      38    1.8%    1.1%       156
13 -    TELEMETRY      35       4     229    1.3%    0.5%        62
```
I am surprised about the day and night difference in the system load (1.87 vs 0.15) and I am surprised about the high values for baro.
This is now just a theoretical discussion. I am happy if someone explains these things to me and anybody else who reads this. But the original problem is found and solved already.

ledvinap, 2016-06-29:
The statistics for baro are incorrect- BMP280 does not use separate temperature measurement, so ut_delay = 0 (https://github.com/cleanflight/cleanflight/blob/058eb5c6f938fc6707a1814e75c5f91d098ee50d/src/main/drivers/barometer_bmp280.c#L123).
Baro UT is scheduled after 100us (sanity limit in scheduler), UP is scheduled after tens of ms (measurement interval).
This is confusing scheduler statistics .

andre92, 2016-06-29:
One thing I notice here is the total task load is about 100% when using BMP280. Maybe when the scheduler has nothing to do it will just check the baro. That is possible since the BMP280 collects all data at once instead of seperate intervals for temperature and pressure. Maybe one interval is not set or zero.

It looks like he BMP280 on the naze rev6 runs as fast as possible, taking the remaining scheduler time. I just tested the BMP085 (on the same board as my external compass) and it did 166Hz with summed avgload of 57.1% and system load of 0.23. 
On my naze full with MS5611 and same settings the baro runs at 99Hz, with summed avgload of 51.9% and system load of 0.27

Ede2016, 2016-06-29:
@ledvinap @andre92 
Would this "incorrect statistic" explain why telemetry and gyro_sync_denom = 1 worked just fine in CleanFlight 12.x but it does not allow me to arm in 13.0?
I have no experience if gyro_sync_denom = 2 is as good as gyro_sync_denom = 1 on my 120mm micro quad (with OneShot KISS ESCs). But it seems with 12.1 it worked and with 13.0 not anymore so I think if it would make sense to go back to 12.1?

ledvinap, 2016-06-29:
@Ede2016 : Only the statistics is incorrect, baro code is working as designed. It is only misleading, baro has minor influence on system load (about 0.5%, 1.061s out of 222s uptime)

System load is average number of tasks waiting during scheduler call. When no task is selected, scheduler is called again immediately (adding 'zero tasks waiting' to average), so 'system load' decreases quickly.

Calculating 'system uptime' / 'total task time' is more reliable way to get actual system load / task cpu percentage.

I'am not sure about difference between 12.x and 13.0, even quite minor change can cause this. Can you try both versions and post task statistics?

Ede2016, 2016-06-30:
@ledvinap 
> I'am not sure about difference between 12.x and 13.0, even quite minor change can cause this. Can you try both versions and post task statistics?

Thanks for the info. Probably I will try this with version 12 and 13 - but probably in a few days. I will publish the information here after I did this.

Ede2016, 2016-07-03:
@ledvinap @andre92 

I changed back to 1.12.1 and here are the status and task details for my AfroMini (Mini Naze32):

All default after flashing 1.12.1
```
# status
System Uptime: 65 seconds, Voltage: 76 * 0.1V (2S battery - OK), System load: 0.15
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP280
Cycle Time: 1006, I2C Errors: 0, config size: 2072

# tasks
Task list:
0 - SYSTEM, max = 8 us, avg = 0 us, total = 4 ms
1 - GYRO/PID, max = 736 us, avg = 642 us, total = 45280 ms
2 - ACCEL, max = 125 us, avg = 93 us, total = 794 ms
3 - SERIAL, max = 635647 us, avg = 2 us, total = 684 ms
4 - BEEPER, max = 8 us, avg = 0 us, total = 12 ms
5 - BATTERY, max = 40079 us, avg = 1 us, total = 67 ms
6 - RX, max = 65 us, avg = 31 us, total = 186 ms
9 - BARO, max = 128 us, avg = 65 us, total = 480 ms
11 - ALTITUDE, max = 270 us, avg = 156 us, total = 441 ms
```
With SBus receiver and SPort Telemetry on SoftSerial:
```
# status
System Uptime: 61 seconds, Voltage: 76 * 0.1V (2S battery - OK), System load: 0.57
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP280
Cycle Time: 1020, I2C Errors: 0, config size: 2072

# tasks
Task list:
0 - SYSTEM, max = 15 us, avg = 0 us, total = 4 ms
1 - GYRO/PID, max = 929 us, avg = 726 us, total = 49503 ms
2 - ACCEL, max = 173 us, avg = 138 us, total = 828 ms
3 - SERIAL, max = 758 us, avg = 2 us, total = 407 ms
4 - BEEPER, max = 12 us, avg = 0 us, total = 12 ms
5 - BATTERY, max = 40081 us, avg = 1 us, total = 74 ms
6 - RX, max = 231 us, avg = 161 us, total = 1120 ms
9 - BARO, max = 145 us, avg = 75 us, total = 427 ms
11 - ALTITUDE, max = 342 us, avg = 180 us, total = 441 ms
13 - TELEMETRY, max = 50 us, avg = 3 us, total = 159 ms
```
Same after set gyro_sync_denom = 2
```
# status
System Uptime: 11 seconds, Voltage: 75 * 0.1V (2S battery - OK), System load: 0.05
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP280
Cycle Time: 1994, I2C Errors: 0, config size: 2072

# tasks
Task list:
0 - SYSTEM, max = 10 us, avg = 0 us, total = 1 ms
1 - GYRO/PID, max = 925 us, avg = 740 us, total = 5170 ms
2 - ACCEL, max = 166 us, avg = 134 us, total = 199 ms
3 - SERIAL, max = 736 us, avg = 2 us, total = 52 ms
4 - BEEPER, max = 10 us, avg = 0 us, total = 2 ms
5 - BATTERY, max = 40088 us, avg = 1 us, total = 46 ms
6 - RX, max = 255 us, avg = 187 us, total = 129 ms
9 - BARO, max = 145 us, avg = 75 us, total = 99 ms
11 - ALTITUDE, max = 360 us, avg = 183 us, total = 116 ms
13 - TELEMETRY, max = 32 us, avg = 1 us, total = 34 ms
```
Change back to gyro_sync_denom = 1, I noticed that system load varies a lot. Here are some lines which I created within 30s:
```
System Uptime: 7 seconds, Voltage: 75 * 0.1V (2S battery - OK), System load: 3.98
System Uptime: 20 seconds, Voltage: 75 * 0.1V (2S battery - OK), System load: 0.84
System Uptime: 24 seconds, Voltage: 75 * 0.1V (2S battery - OK), System load: 0.77
System Uptime: 25 seconds, Voltage: 76 * 0.1V (2S battery - OK), System load: 0.73
System Uptime: 26 seconds, Voltage: 75 * 0.1V (2S battery - OK), System load: 0.77
System Uptime: 27 seconds, Voltage: 75 * 0.1V (2S battery - OK), System load: 0.82
System Uptime: 28 seconds, Voltage: 75 * 0.1V (2S battery - OK), System load: 0.55
System Uptime: 29 seconds, Voltage: 76 * 0.1V (2S battery - OK), System load: 0.71
System Uptime: 30 seconds, Voltage: 75 * 0.1V (2S battery - OK), System load: 0.79

```
I will stay with CF 1.12.1 for the moment and look what the future will bring.
Here is some more info about the same issue with some more details:
http://www.rcgroups.com/forums/showthread.php?t=2678624

closed 2017-02-25 by hydra