cleanflight/cleanflight #2259: Aux2 channel 6 not working on CC3D  or CC3D ATOM

codeMonkeyDo, 2016-06-25:
I have installed cleanflight onto a standard CC3D FC board (at version 1.10 using the .bin) and a CC3D Atom FC (at version 1.13 using the .hex). On both occasions I noticed that I could not get any input from channel 6 which I normally would see as AUX2. 

When I first noticed this I ignored it assuming that perhaps it was an issue with my wiring or the setup in general but since repeating the process a second time on a completely different installation with different Rx/Tx units and seeing it again I have to assume that it is a bug.

Z-slim, 2016-06-27:
Are you using PWM RX? was it the same board in the other installations (as in moved between builds, and not just a second CC3D) or are several different CC3D boards doing the same? Have they ever been connected to the same hardware before, like same physical rx moved between builds or tested FCs with? Lastly, with different TX/RX combos... same setups? (brand of units vs just using a different unit from same mfg, etc..)  

I had an input issue with PWM and it turned out to be a burnt signal line on the CC3D due to a faulty rx. Couldn't figure it out until same rx unit was pulled and used on another CC3d and presto, lost a channel on that one too! I had tried all manner of HW combos, but the CC3ds remained the same... sucks a burnt 2 to find out it was a bad RX... CH5 was doing something funky and not very servo-signal-like on the o'scope when I clued in. My APM board survived that RX however, but I wasn't even concerned with ch5 for that build and never noticed it. Board is still working on ch5 in a different craft now.

MrJCub, 2016-06-27:
If you are using PWM, then you only get a maximum of 5 input channels on the CC3D.
See the manual:
[image]

Z-slim, 2016-06-28:
Ah, my bad, I missed that one. I only used PWM in the very beginning and results above made me switch to PPM and then direct Sat., so I guess I assumed 6 inputs in CF because OpenPilot had it. I have a cc3d in my 250 racer with CF 1.13, on satellite rx and all 12 channels are working straight away, so if not PWM, I'm not sure what could be the problem aside from computer used to flash from? Maybe FTDI or USB weird issue? Try different ones, can't hurt. I've been suckered by taking certain things like those for granted and thinking the problem lies within my circuitry design or mcu code. Granted, PICs are different beasts, but have seen the improbable happen due to dodgy burning setup. I haven't played with these chips yet, so that's all I've got. Best of luck. Z.

MrJCub, 2016-06-29:
I'm yet to switch to PPM and the 5 channels did catch me out when I assumed also the inputs were same as Openpilot/Librepilot.

codeMonkeyDo, 2016-07-11:
@Z-slim - This was with two completely different quad copter setups using different CC3D FC boards and different Rx's. 

Yes - I am using PWM at the moment as I was simply reconfiguring from LibrePilot to Cleanflight. 

@MrJCub - I guess you have answered this issue. Using PWM on CC3D only supports 5 channels. 

Does Cleanflight support PPM or SBUS/IBUS on CC3D?

MrJCub, 2016-07-11:
Yes, Cleanflight supports PPM on CC3D, It's pin 8. See manual here: https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20CC3D.md
When using PPM, you get all input channels and also can use more features, like voltage and current sensors as well as LED strips. In the manual, the first two tables are for using PPM and the second two are for using PWM. (I originally didn't read it correctly and thought I could use LED strip on PWM, but then noticed it only works with PPM).

closed 2017-02-25 by hydra