cleanflight/cleanflight #2260: SPR Evo: Access the SD Card though the configurator

soler2000, 2016-06-26:
Currently the cleanflight configurator does not have the ability to access the SD card and the intention is to remove the card and plug into the PC to view the logs.

In some craft if it difficult to access the card (on my gravity 250 I need to open the frame and remove the FC to access it).  It would be good to be able to down load the logs (all or some) from the configurator to save disassembling the craft each time,  even if this is at a much slower data transfer speed.

Linjieqiang, 2016-06-26:
It sounds great~

ledvinap, 2016-06-26:
It will be easy to modify FLASHFS commands to support raw SDcard access.

FAT implementation may be in configurator0; is there any javascript FAT implementation? How hard is it?

closed 2017-02-25 by hydra