cleanflight/cleanflight #2265: Is AK8975 compass still supported?

krzysztofmatula, 2016-06-27:
I'm experimenting with an external CJMCU-86 module and the AK8975 compass. I'm connecting the module over I2C to my SPRACINGF3 deluxe.

The issue is that whichever mag_hardware I use (0, 2, 3, 4) it seems always the on-board compass is in use.

I didn't verify if my AK8975 board is actually working, but I'd expect the "heading" to show some default value (zero, as in case of mag_hardware=1) after setting mag_hardware=3 in case of failure. Instead it shows the readings from on-board HMC5883.

Actually, it can be tested without the AK8975 module. I'd expect the board would behave as if no mag is available when using mag_hardware=3 in such case.

Tested on v1.13.0.

andre92, 2016-06-30:
AK8975 should be supported on SPRACINGF3 target (not mini or evo) It is defined in cleanflight/src/main/target/SPRACINGF3/target.h line 54: #define USE_MAG_AK8975
Please note: not all targets have this mag defined, but since it is i2c, it is no problem to add it for your own board and rebuild.

Actually, if you specify a non-default mag (mag_hardware >= 2) and the board fails to detect it, it will use the default (onboard) mag, without changing mag_hardware, since that setting is only used at startup.

krzysztofmatula, 2016-07-08:
Thanks, I was not aware of the fallback-to-default process.

closed 2016-07-08 by krzysztofmatula