cleanflight/cleanflight #2266: Lack of Multiwii Serial Protocol (MWSP) support?

agentrfr, 2016-06-28:
Enabling serial comms on current version 1.13 allows some data to be read over serial (specifically attitude and current rx settings), but does not support writing writing rx values (for example setting roll, pitch, yaw, throttle and aux channels).

I've tried writing 8 byte, 16 byte and 32 byte packets to the board, but it refuses to set rx values.

Tried on baseflight, writing rx works perfectly with 16 byte packets :\

Please help if I'm missing something blatently obvious...

andre92, 2016-06-28:
Did you set receiver mode to rx_msp in the configurator?

agentrfr, 2016-06-28:
Yip enabled rx_msp in the CLI. Doesnt let me set rx values. I can read them, not set them.

I mentioned 1.13 because it is the one I'm on and didnt want any confusion, havent tried any earlier versions as it's my first time playing with a naze32.

I've had a quick look through the code and can't see anything related to an safety "feature" to arm rx reception from serial.

I know the board is getting the packets becuse I don't receive an error and the packet checksums check out.

ledvinap, 2016-06-28:
Tested on SPARKY, in CF configurator:
- Flash 13.0
- set receiver mode to RX_MSP
- on receiver tab, enable control sticks

Displayed RX channels match control sticks .. 

Can you confirm that this works for you?

conkerkh, 2016-06-28:
It works and you can transmit RX through MSP, the only problem is that there is smoothing now enabled for all RX recovers and since RX_MSP is completely proprietary implementation there is no refresh rate on this receiver, either set it in code or modify void processRcCommand(void) in mw.c so it will report command without smoothing for receivers without refresh rate. (personally i prefer second option for now).

ledvinap, 2016-06-29:
@conkerkh: #2183 shall fix refresh rate issue, it is merged in master

andre92, 2016-06-29:
@ledvinap It works, except throttle and yaw channels are switched out for me. Keeps working in all tabs except CLI.

Benik3, 2016-07-12:
I use msp_RX with EZ-GUI and it's working.
But it has delay. Is it because of the mobile app side or because of the board side?
(I use ESP8266 as receiver)

conkerkh, 2016-07-19:
@Benik3 Most likely their firmware that they put on ESP has some stupid checks and lags. I personally also used UDP instead of TCP and had much better results: 0 lag quad was flying well. Anyway you are really brave to fly quadcopter with your phone you don't have feedback at all that you usually have with some sort of physical controller, it makes everything difficult.

Benik3, 2016-07-19:
@conkerkh Thanks for reply.
Well I use it with CJMCU board (10cm mini quad). I wouldn't never use it with bigger quad :)
I use FW from jeelabs. How do you made the UDP? (you can send me PM so we will not mess this thread...)
Thank you

conkerkh, 2016-07-19:
@Benik3 I used custom stuff where Raspberry was acting as ground controller sending custom UDP packets and ESP was processing them. I've written firmware for ESP with Arduino but it was using ESP SDK functions so everything works properly. It's a bit hard to explain it exactly but I don't think EZ-GUI can work with this solution you would have to write everything from scratch.

Benik3, 2016-07-19:
@conkerkh Do you think, that bluetooth modules are faster?

conkerkh, 2016-07-19:
@Benik3 it doesn't depend on module but on implementation if the app is badly coded it will work bad. Bluetooth might work better in this case cause module comes loaded with serial service but it has terrible range

Niko-La, 2016-10-11:
Is naze32 still unable to send MSP_SET_RAW_RC CODE.  I am able to read ATITUDE but my MSP motor commands are not going through.

My python script was sending commands on a MultiWii PirateNG fine.  Just replaced with Naze32 with Cleanfilight.

Can anyone confirm if they Beta/Base/Racerflight firmware are working with MSP_SET_RAW_RC CODE.

@conkerkh Hey can you please elborate on your method to fix this?

conkerkh, 2016-10-12:
Set your receiver for RX_MSP in cleanflight configurator.

calvinl-l-l, 2016-11-24:
Hi, sorry to drag on this issue. But I have a question about MSP_SET_RAW_RC and coudnt find it anywhere else. 

I am now able to use MSP_SET_RAW_RC to control the FC via MSP on my computer. However, after I set RX_MSP, the legacy RX no longer works (cant control via RC transmitter), as receiver mode is required to be set to RX_SERIAL to use sbus. So my question is, is it possible to have the FC listen to the MSP rc override and the sbus at the same time. For example, I want to only override the pitch channel with MSP_SET_RAW_RC and leave the rest of the channels to be control by sbus signal.

If above isnt possible, I find that after I send MSP_SET_RAW_RC, then if I try to read MSP_RC, it will return the overridden values. Is there a way I can listen to the signal from the sbus (I could piggyback the sbus wire on an arduino and read the signal, but I prefer not to), and maybe just forward those value using rc override. Then I can have control of which channel I wish to override. 

Thanks

Niko-La, 2016-11-29:
@calvinl-l-l

 Hi are you using python to send MSP_SET_RAW_PC? Is so can you provide the code please.

@conkerkh I am able to send control but for some reason my quad keeps yawing when i give it  throttle for lift off.

At the MSP (http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol)

MSP_RC	105	FC →	rcData[RC_CHANS]	16 x UINT 16	Range [1000;2000]
ROLL/PITCH/YAW/THROTTLE/AUX1/AUX2/AUX3AUX4

1) Is there / missing between AUX3AUX4?

2) here is my sample data i am sending
    data = [ROL,PIT,YAW,THR,1800,1500,1500,1500]

I am getting motors to spin when moving rol,pit, yaw and thr.  but when i put on propellors, the quad just keeps yawning  CW>

calvinl-l-l, 2016-12-02:
@Niko-La. sorry all my code is actually written in c++, but I did come across this [https://github.com/alduxvm/DronePilot](url), maybe you can try this and see if it works for you.

conkerkh, 2016-12-21:
@Niko-La I've written my Python library for MSP with threading and all goodies and everything works nicely. Although there are few hiccups here and there so I'm not gonna release it at the moment. There is something similar to what I've done on GitHub though I haven't gone through the code that much.

closed 2017-02-25 by hydra