cleanflight/cleanflight #2267: Can't get more than 6 auxiliary channels

Prowlr, 2016-06-28:
I'm using the Taranis with the X4R-SB and I can only assign 6 aux channels. Even though I have channel 7 and 8 setup on my transmitter, cleanflight doesn't seem to be receiving the signal from the rest of the aux channels. The channel range on the Taranis is set to 'CH1 - 16'. Not sure why I'm limited to only 6 aux channels. Any ideas?

GuybrushX, 2016-07-01:
Input/Mixer settings wrong on your Taranis?

I'm using a Tanaris + X4R-SB, too and I can see and control all 16 channels configured via SBus.

Eachine Racer 250 here with Cleanflight 1.13.0, Betaflight 2.9.0, Librepilot 15.09, all working.

Seems more like a configuration issue instead of a bug.

Prowlr, 2016-07-01:
Well, I have the channels that don't work set up the same way as the ones that do work.

hydra, 2016-08-10:
configuration issue with TX.  if you can receive SBus input for 1 channel from the X4RSB then the hardware is OK.  Suggest watching some videos online or asking on opentx forums for help.

closed 2016-08-10 by hydra