cleanflight/cleanflight #2270: Bug - Can't arm when using luxfloat

gegtor, 2016-06-30:
when i enable luxfloat on my naze32 i can't arm because of arming angle is exceeded (green light blink)

i tried to calibrate acc and gyro and mag

andre92, 2016-06-30:
Naze32 does not have a floating point unit, which means it can not do floating point calculations as fast as most F3 boards. You are overloading the cpu by running luxfloat on a Naze with all other settings default. My system load went from 0.6 to 2.0 just by switching from rewrite to luxfloat. Use rewrite instead, it uses the same pid values as luxfloat. 
Use the 'status' command in cli to see your system load. 1.0 or higher and it won't arm.
Green blinking led only indicates it won't arm, not why. Failsafe also makes it blink.

If you really want to use luxfloat on Naze, you can either increase looptime to 2000 by setting gyro_sync_denom = 2, or disable some features/ sensors to free up processor time.

gegtor, 2016-06-30:
Thanks i see :)

closed 2016-06-30 by gegtor