cleanflight/cleanflight #2275: No HOTT Telemetry an V1.13

JocB1, 2016-07-02:
I used HOTT telemetry on the old versions including 1.12.1 without problems. With 1.13 HOTT is not working, I used first the old working setiings (restore from 1.12.1) then I tried UART1, Softser1, Softser2.

hydra, 2016-07-04:
Funny you should say this as yesterday I installed 1.13.0 on my SPRacingF3 Deluxe in my old TBS Discovery yesterday and GPS and ELECTRIC air were detected by my MX20 and worked just fine.

JocB1, 2016-07-05:
I use a Flip32 and a Naze32, both with the same behavior. The receivers are GR-12 and GR-12s, the newest firmware was already istalled. On my MX-20 I updated yesterday to the newest 2.x4 to 2.x7 version without change. The HOTT wireing is exactly as in the docu with 1N4148 diode. I tried also a full chip erase and set up the config rootedly new, without success. But one thing was confusing: Directly after flashing from 1.12.1 to 1.13.0 (without interrupt the power) and backing up the old settings the telemetry was working with 13.0 shortly till I reconnect the battery. One thing I have to test a bit better, but I have the impersion: When I connect a GPS module to 1.12.1 then HOTT stopps also working.

janlang, 2016-07-07:
Hi there, I have similar/same topic. I have also problems with HoTT protocol. With the newer MX Graupner software versions the transmitter only requests the different modules connected to the receiver for lets say 30s. After that he will not recognize the modules anymore. the general electric air module works at my configuration with the newest software. but when i add the GPS none of the modules is shown anymore. Setup is as following:
MX-12 with a GR-12l
GR-12l (SUMD) --> UART2
GPS --> Soft serial 1
Telemetry --> Soft serial 2

GPS is getting a fix and is shown in the GUI when activated. Telemetry is working as long as no GPS is added.

janlang, 2016-07-07:
One more to add. Maybe it would be interesting when the software starts to broadcast the HoTT data packages. It would be best if it starts directly after startup even if no data is available at that point...

MarkusAndreasMueller, 2016-07-09:
Hi guys, have the same problem.
I've used up to now the Graupner receiver as FC. But now me I bought a SP racing F3 Deluxe.

I'm using:
-Cleanflight version 1.13
-Graupner GR16 version 6.37 (SUMD on channel 8 - works fine)
-Graupner MC-32 with the version of 1.083
-no GPS

I've tried the following:
-Diode installed as described in the instructions.
-UART 2 with hott telemetry, with different baudrates - dosn't work.
-SoftSerial 1 and 2 with hott telemetry with different baudrates - dosn't work.

I got me the level with my old analog oscilloscope viewed (SoftSerial), the levels seem to be ok (~ 3V3).) But I suspect a problem with the Bautrate. Unfortunately, I could not properly measure them with my old device.
I have observed that the GR16 cyclically sends data (~ 1 second).

janlang, 2016-07-09:
Hi, telemetry works in general. Please check your connection and first start the copter and then the sender. The sequence is crucial, because the sender checks for the receiver sensors only for a few seconds. After that press the < button on the sender and then the down button which leads to the telemetry selection menu. Did you already activate the telemetry pin in your receiver by setting it to on in the sender?

MarkusAndreasMueller, 2016-07-30:
I just did it how you wrote it. But I can't activate the sensors in the transmitter, the recognized sensors will only shown me. And it will be shown only the receiver.
Also I tried with GR32 receiver and with the latest software for my MC 32 - without success.

rvmicide, 2016-09-18:
Hi, 

I have the same with my naze32, mx20 Hott radio and GR-24 receiver. No telemetry with cleanflight 1.13.
If i flash betaflight 2.9.1 that's work (after activate softserial, telemetry, and hott protocol) But when i return in cleanflight and activate the same option, the transmitter don't detect air module (same hardware, cabling, and procedure, config....)

Like JocB1, when i go to CF 1.13 from BF 2.9.1 the air module is visible in the tx for a few minutes but i feel is just the telemetry menu is not refreshed.

closed 2017-02-25 by hydra