cleanflight/cleanflight #2279: USARTs and UARTs

martinbudden, 2016-07-05:
@hydra ,

a while back you changed some of the `USART` `#defines` to `UART`. As I understand it some boards have USARTs and some have UARTs, but Cleanflight itself only ever uses serial ports asynchronously (ie UARTs). What is your view on how the terms UART and USART should be used within `#defines` and variable names so that the terms are used:

1. consistently
2. in a meaningful way

I ask because changes are being done in this area in betaflight and these changes are likely to make there way back into Cleanflight (at least in an indirect way) when Cleanflight moves over to using the improved `make` system and also when Cleanflight moves to the new IO and also when Cleanflight does the F4 port.

So it would be good to ensure the changes in betaflight are consistent with your views on how the terms should be used.

hydra, 2016-07-05:
yes, i used UART since we don't use the syncronous/clock generation features of the ports.  Also UART4/5 on F3 boards are not USART ports.  They are UART only so it made sense at that time to fix it.

I have no strict views on it.

martinbudden, 2016-07-14:
Closing as question answered.

closed 2016-07-15 by martinbudden