cleanflight/cleanflight #2280: Optimized buffer handling

ledvinap, 2016-07-05:
I was thinking about extending current streamBuf implementation, allowing direct use of circular buffers and optimization of some functions

- augment sbuf_t structure with 'refresh'/'virtual' function pointer. 
 
 All space after this will be driver-specific. Refresh function can store necessary internal data here, similar to serial port 'virtual' interface

 This (virtual) function will replace `*ptr` and `*end` with new buffer if old one is exhausted and will return total available space (both for reading and writing).

 Maybe more functions (virtual table) may be used (but I would prefer argument to differentiate call type if needed)

- add macro to check and conditionally refresh buffer
This will improve generated code - compiler may keep `ptr` in register, refreshing it only when new buffer is used. Also current barier behavior of serial function (function called through pointer may modify any non-local memory, compiler needs to reload all cached variables) will be improved.

- add small (~8 bytes) reserve space after each buffer (buffer chunk)

 This way, there is always guarantied space available in buffer, so function that write or read small amount of data don't need explicit check (or check after each byte). 
 Serializing fixed size chunks will be easier (MSP serialize/deserialize) and some blackbox serialization will benefit too (Variable byte encoding should fit in - currently at most 5 bytes).
 Space check may be even moved to user level (MSP messages), with both checking and non-checking variants supported. 
 This 'overflow' space will be copied to next buffer in refresh function for write streamBuf and mirrored from next buffer for read.

- implement streamBuf interface for serial port 
 This will greatly improve performance - no blind calling through pointer for each byte, with overflow and boundary checks.
 Two functions may be used - one to get available buffer (without actually allocating/modifying serial state) and another to commit streamBuf (adjusting pointers to write data / freeing data that were read). 
This will allow modifying message header when message is finished - mostly message lengths (MSP etc) or computing checksum of message. Only requirement is that message must fit in serial buffer

- implement streamBuf interface for printf
 Benefits will be similar to serial port case (but using write instead putc semantics in printf will have similar effects)

- implement streamBuf interface for flashFS and FAT
 This should be quite trivial, flashFS already uses similar interface internally and FAT has large linear buffers

Some chained-block based allocator may be implemented in future (single-linked-list of message blocks), without changing streamBuf interface. 

What do you think? Is it worth the effort?

ledvinap, 2016-07-05:
BTW: I got to this when I tried converting Blackbox to memory-buffer based format. Sometime multiple Blackbox frames generated, which makes it a bit difficult to separate buffer allocation from rest of code. This streamBuf approach will make it a bit simpler ...

martinbudden, 2016-07-06:
These are certainly beneficial changes. I've used a system before where there is reserved space after each buffer and it certainly worked well.

As to "is it worth the effort", I can't really say. I would say it partly boils down to whether there is something else of more worth you could spend your time on.

hydra, 2016-07-13:
@ledvinap I agree with @martinbudden 's thoughts.  Perhaps there is something more important.  I see merging the IO changes that you started that have been picked up by betaflight/raceflight back into cleanflight as being far more beneficial at this time.

That said, the change will likely be quite isolated with minimal impact.  I'll let you decide.  I think the 8 byte chunk/reserve space and the serial port changes will likely be the most noticeable changes and are definitely worth the effort.

closed 2017-02-25 by hydra