cleanflight/cleanflight #2281: Feature request: support configurating VTX via Stick commands

ps915, 2016-07-05:
What about to add a new "standard" for configuration a vtx with an data wire to the naze32 or any other FC.

Maybe the TBS Unify Pro would be a good start. It would be awesome if i can configure it via cleanflight.

For example you your quad has to be disarmed and you need  to switch an AUX Channel to activate the configuration for the Unify Pro. Then you can go with your sticks through the menu of the unify.

The other thing is, do generate a protocol to generally configure VTXs. So maybe in the future we do not need any dipswitches, leds or segmentdisplays for configurating the videoTx

the guys from dRonin just did that for the unify pro: https://github.com/d-ronin/dRonin/pull/1067

What do you think?

cheers,
Phil

vadav, 2016-07-05:
Would be really nice!!

jaimevisser, 2016-07-22:
See also:

https://github.com/betaflight/betaflight/issues/809

rdarby777, 2016-08-17:
Is anyone currently working on this feature?

closed 2017-02-25 by hydra