cleanflight/cleanflight #2282: Flying issue!  PLEASE HELP!

quadguy, 2016-07-05:
I just built my first quad and im trying to finish this quad so i could start having some fun.  The issue is the quad cant re-center  it's self.  If i fly  forward and let go of the pitch the quad will still flying full pitch unless i pull the pitch back.  I have the same issue with roll as well.  I roll left and let go the quad still will be in roll until i roll the other direction.  im not sure if it is an issue with the receiver,fc,or PID.  Ive been playing with the PID and still cant get the quad to level after transmitter commands.

daPhoosa, 2016-07-05:
You're vey likely in Acro mode.  The best thing for you to do would be to look up Joshua Bardwell on YouTube and watch his series on learning to fly a quad.

The less ideal alternately is that you can map one of you aux channels to turn on either Horizon or angle mode, both of which will auto level once stick return to center.

Good Luck.

quadguy, 2016-07-05:
ill look that up and try it out.  Thanks daPhoosa.  Do you happen to have a link to the video?

quadguy, 2016-07-05:
I looked under modes and dont see Acro mode?  I have not played with any of the modes in mode section.  Do i need to work with the modes so i can fly with auto center?

martinbudden, 2016-07-05:
@quadguy , the default mode is Acro mode. You need to change the settings so that either angle or horizon mode is on. The easiest way to do that is to set the whole range for one of the aux channels to be angle mode.

(Angle mode is what you call "auto center").

quadguy, 2016-07-05:
do i move the bar all the way up? going to the right?

quadguy, 2016-07-05:
IT WORKS!  i figured out how to change the mode and she flies great.  something very simple but not to a newbie.  Ive been stuck on this for a week.   Thanks to the members that participated.  Time to go out and have some fun and do some fine tunning.

closed 2016-07-05 by quadguy