cleanflight/cleanflight #2290: bug report: possible magnetometer bug

samerhay, 2016-07-08:
NAZE32 with 10DOF and GPS.
Magnetometer declination is -6.5.
everything was working fine.
I modified the arrangement of the receiver and the GPS inside the quadcopter, which forced me to rotate the flight controller 90 degrees. I compensated for that in the "configuration - board and sensor alignment". the configuration is working fine as  determined by looking at the virtual drone on my screen in the "setup" tab (what i mean, if you lift the nose, the right movement is detected and if you roll left, it roll left, and if yaw right, it yaws the right direction). But now comes the problem: my true north is pointing east. I recalibrated the magnetometer. it was still wrong. the only thing that fixed it, is when i deleted the magnetometer declination value, rebooted, put it back, and reboot again.

hydra, 2016-07-13:
thanks for the bug report, we'll look at the potential fix.

hydra, 2016-07-13:
fix merged, please test any development build from tomorrow.

Download one from here:

http://cleanflight.memoryleaks.org/builds/

PR #2300 (2016-07-13): don't scale mag_declination by 10 (readd)
This change was added in cleanflight and then got left behind in a cleanup.
This brings an incompatiblity with the configurator.
Possibly fixes  #2290
closed 2016-07-13 by hydra