cleanflight/cleanflight #2292: error seeking dterm

andre92, 2016-07-09:
It looks like the Dterm behaviour was changed permanently to MEASUREMENT in 1.13, without the possibility to set it to ERROR. (which was default in MWrewrite before 1.12)
Is it possible to get the 1.12 behaviour back where users can choose the behaviour?
The problem is for copters that need to use agressive filtering on the Delta. (big noisy copter here, below 20hz) 
The Dterm lag causes bounce backs after flips and rolls (Delta wants to continue a move while P and I try to stop) which you cannot tune out with such agressive Dterm filtering (since in MEASUREMENT Dterm only opposes P and I terms)
 If the Dterm was error seeking, it would instead help the P and I to stop the move (instead of try to continue it) and it would give more response and less bounce back.

martinbudden, 2016-07-09:
@andre92 ,

Derivative by measurement and derivative by error are equivalent, except for a "kick" when the Set Point changes. See http://controlguru.com/pid-control-and-derivative-on-measurement/ for an explanation.

There is an argument for allowing the choice of derivative from error, because some pilots see this "kick" as useful.

As part of the 1.14 changes, I'm planning to add back the ability to choose derivative by measurement or error, however I don't think this will help solve your problem.

I think you have your DTerm filtering set **way** to aggressively. I think you'd probably get a better tune by setting the filter at something like 70Hz and lowering your D value.

andre92, 2016-07-09:
I think you are right about my filtering. I made a blackbox log and concluded 2 things:
1. my Dterm lags too much due to low cutoff frequency, and my D gains were too high. (I'll double frequency, reduce gains by half and see how that works)
2. I have way too much gyro noise, so I will first soft mount my Naze rev6 and then try again for Dterm filter. I hope less noise will also improve flight times since there will be less jitter in the motors.

I'm indeed one of those pilots who see the "kick" as useful, because it intends to reduce overshoot and bounce back (especially when stopping after a flip or roll). It does feel more robotic but at the same time more locked in. 
If you compare a blackbox log for both error and measurement, you'll see with error all 3 terms work in the same direction, whereas with measurement, derivative is mostly opposing P, reducing the response if set too high.

borisbstyle, 2016-07-09:
Derivative on measurement actually gives you less bounceback generally if
thats what you are wanting to solve?

andre92, 2016-07-09:
@borisbstyle after some reading and watching some videos, I concluded my dterm lpf was way too low. I previously thought it was ok because CLI let me set it. What is the lowest setting you would recommend on a very noisy copter? (500 X with plastic molded arms)

hydra, 2016-07-13:
ok, i think we can close this.  if @martinbudden does a PR as he mentions we can reference this issue when he does.  @andre92 good luck with your tuning.

@borisbstyle and @martinbudden thanks for comments.

closed 2016-07-13 by hydra