cleanflight/cleanflight #2294: Defect bitmasking in smartport telemetry

tedbarth, 2016-07-09:
I guess I found a bug causing telemetry modes at the wrong bit. At least one of them should add `+= 80` instead of `+=40`. I know UNUSED_MODE is not used but in future it could lead to a wrong bit masking.

https://github.com/cleanflight/cleanflight/blob/master/src/main/telemetry/smartport.c#L406

```
                if (FLIGHT_MODE(UNUSED_MODE))
                    tmpi += 40;
                if (FLIGHT_MODE(PASSTHRU_MODE))
                    tmpi += 40;```

tedbarth, 2016-07-09:
BTW: Can you give me a pseudo code example on how to read if a special mode (eg. horizon) is set given a sum?

ledvinap, 2016-07-09:
You can't add 80 - it won't fit in single (decimal) digit.
This code is for displaying information only; just deleting UNUSED_MODE here is probably OK

tedbarth, 2016-07-09:
So you would never know if UNUSED_MODE or PASSTHRU_MODE is actived, only both or none. Right?

Found a way to check a flag:

     sonar_mode = (SONAR_MODE / 100) & 4) > 0;

hydra, 2016-07-13:
code cleaned up in 2507b90a2827496fd9aae04624bc2a8aabbb64f3

nothing more to do here.

closed 2016-07-13 by hydra