cleanflight/cleanflight #2296: GPS softserial dropouts

kaefert, 2016-07-11:
connecting the GPS to a softserial port on my flip32 (naze32 clone) makes the gps connection dropout for about 2 seconds every 10 seconds. (visible to me on my minimOSD).

When I connect it to an UART it works much better no dropouts and much faster updates, but I only have 2 UART ports and I need one for the OSD and the other for the blackboxlogger.

I do believe that in previous cleanflight versions these dropouts where not there, the speed was always slower (softserial vs uart) but I thnk in the past there where no dropouts in between

andre92, 2016-07-11:
Without more information it is hard to tell. May I see a cli dump, and the output of "status" and "tasks"? I am interested in your gps settings, baud rates, looptime and cpu load.

I am not sure issue 870 is relevant here since that user had an issue with the gps module itself.
If your gps runs fine on uart but not on softserial, the issue must be something in your softserial.
Have you tried running the gps on softserial at different baud rates?

kaefert, 2016-07-11:
Hey there.
I'm sorry but I don't have my quadcopter with me, I'll only get to it next weekend so friday the 15th july (in the evening)
So I can't give you a CLI dump right now.

But I can tell you that I did try different baud rates but it does only work on the highest setting, I guess I would need to somehow also tell the gps module to communicate on a different baud rate, without that changing only the baud rate in cleanflight creates a mismatch that prevents the flightcontroller & the gps module from understanding each other?

I can give you my last blackbox log file (with gps on softserial, since uart is used for blackbox logger) if that helps?

andre92, 2016-07-11:
Sure, anything that helps, maybe I can view the gps data info from your blackbox log. What kind of gps do you have, is it ublox or nmea?

kaefert, 2016-07-11:
It's a ublox module:
http://www.banggood.com/UBLOX-NEO-M8N-BN-880-Flight-Control-GPS-Module-Dual-Module-Compass-p-971082.html

my last Blackbox log file:
[LOG00329.TXT](https://github.com/cleanflight/cleanflight/files/357588/LOG00329.TXT)

andre92, 2016-07-11:
I see 2 moments during flight where you lose gps fix. Did you configure your gps module with U-center? I see the default baud is 38400 and it supports all baud rates. I'm still not sure whether the gps is set to output ublox or nmea protocol, it is not in the blackbox file. Some ublox gps modules output nmea by default, which uses a lot of bandwidth, which increases the risk of error.

By the way, were you tuning pids when you made this log? d_roll looks a bit high to me.

kaefert, 2016-07-12:
only 2? hmm, maybe that was a short flight, I will send you a previous longer flight than I'm sure there would be more dropouts...

No I did not configure the gps module, at all. Just using it as it came from the chinese seller..

I tried tuning pids a few times in the past, but I never managed to get any results that felt better than the default settings. The only useful change I found was to make the yaw axes much more responsive, with default settings I always had to max out that axis on my remote to get a painfully slow yaw turn.

jflyper, 2016-07-12:
38400 is too fast:
[https://github.com/cleanflight/cleanflight/blob/master/docs/Serial.md](url)
> **Constraints**
- Softserial is limited to 19200 baud.

andre92, 2016-07-12:
@jflyper strange, I manage to run both gps and smartport telemetry at 57600 baud on softserial and it runs fine. I could even get my looptime down to 1500 with gyro_lpf = off and gyro_sync_denom = 12
I only had to disable nmea output on my gps module (some ublox output nmea by default until you disable it in u-center)

jflyper, 2016-07-12:
@andre92 The constraint was probably set with continuous input/output streams in mind.

Nevertheless, there are no reason to push things out of spec when the system has a problem.

kaefert, 2016-07-12:
is there some way to configure a gps module without a Microsoft Windows PC? As far as I found this "U-Center" software is only available for windows?

jflyper, 2016-07-12:
@kaefert I've never done this before, but there is a code that changes the baud rate according the the FC's configuration. Worth trying?

kaefert, 2016-07-12:
could you elaborate? what do you mean by "there is a code" (that sounds kind of mystical ;) )
But I guess I could just try this u-center software on a friends Windows PC or using wine next weekend

jflyper, 2016-07-12:
The current CleanFlight looked to me that it already had a code (functionality) to re-configure the GPS's sending speed, but ...

Daaaahhh, I took a second look now, and it only works for the UBX mode. You have to switch to UBX with U-center anyway to use this function.

hydra, 2016-07-13:
there are already PR's that look to improve softserial performance that have not been merged due to incomplete code and time.  Use a hardware UART for GPS please.  Softserial is meant for low-speed and low bandwidth telemetry protocols.

closed 2016-07-13 by hydra

kaefert, 2016-07-16:
Thanks everybody for your help, funny enough I just updated my flip32 from version 1.12.1 (which showed the described problem for me) to version 1.13.0 and with that I was able to set the serial speed for gps to a lower setting and the update rate of the gps-coordinates on my OSD is now just as high with the softserial as it was previously only connected to a hardware UART slot