cleanflight/cleanflight #2299: Control Cleanflight FC through l2c port using RPi or some other micro processor

dantheman213, 2016-07-12:
There should be a way for Cleanflight to give out sensor data and information to a requesting device from the l2c port. More advanced than the MSP protocol.

It would be amazing if the outputs included all sensor data and you could also issues GPS waypoint commands or other more refined controls so that a device like the Raspberry Pi could easily give direction to a flight controller. 

However, since the flight controller is supposed to take care of things like PID and actual control motors and thrust, it would be great to a have a more simplified access to these resources without having to control everything yourself through the MSP protocol like now. I should be able to say "go in this direction" rather than say "spin up motors 1 3 4 and use this throttle value". That's what the FC is for!

This would allow for more autonomous UAS as well as cooler projects that would allow for more experimental RC vehicles.

ledvinap, 2016-07-13:
I2C is probably too slow for this - you will eventually want to transfer all sensor data with minimum latency. Serial line should work fine (and it is usually well supported under linux)

It may be reasonable to keep only low-level ACRO PID + failsafe/override on FC and move other parts to (much faster) parent processor. Control task would be easier to interface than processor .. 

Intel Edison seems like interesting choice due to it's form factor. It is quite easy to build RTLinux image, with latenciy in us range.

dantheman213, 2016-07-13:
Since the FC was designed to be responsible for all core flight operations, wouldn't it make sense to keep all flight operations on the FC and just provide a simplified API for external devices to connect via serial or whatever communication port?

hydra, 2016-07-13:
I'm not sure this is a good direction or core to cleanflight's goals.  Yes we could do it, I don't have time to review any PR's that implement such code, there's way more pressing things that need to be done.  Feel free to implement it yourself and then post a link to your code from this issue for reference though.

closed 2016-07-13 by hydra

lewma, 2016-09-01:
I agree with @dantheman213.  It would be great to have the FC just concentrate on controlling flight and have an API for a secondary processor ( like a RPi ) to send it intelligent commands.

I also agree that this is probably not the direction that cleanflight should be going ;)

If I get enough time, perhaps I'll fork and see where it goes.