cleanflight/cleanflight #2302: Parameter Group over MSP

ledvinap, 2016-07-13:
PG over MSP+configurator support should be implemented. I'am opening this to discuss implementation/coordinate effort.

Is someone working on this ?

hydra, 2016-07-13:
I'm not currently working on it.

yes we need it for backup/restore so it would be awesome if you have time.

ledvinap, 2016-07-18:
Here are some thoughts:

MSP/CF part is quite simple:
- raw parameter group transfer - pgn ; version ; length ; binary data
 - both read and write, sanitize length on write
 - arrays can be transferred by index, item by item (this is a bit complicated - only total array size is stored in `pg` descriptor; this should be fixed anyway)
 - what about Profile? Assume access only to selected profile and handle it implicitly?
- transfer info from `pgRegistry_t` - pgn ; version ; record size ; [item size] ; system/profile flag
 - simple message mapping internal (packed) representation should be enough
 - some record indexing is needed. Simple indexing (record index as stored in memory), iterating records for each operation may be enough. 
 Or use indexing by PGN + implement list of known PGNs

This should be enough to list complete configuration, modify it, do non-trivial version migration etc.

ledvinap, 2016-07-18:
The configurator part is more interesting

My idea is to implement mechanism to describe all pgns including different versions. This can be used to map binary representation from CF to javascript object.

Implementation of new configs will be easy - just describe new configuration as C structure and JS descriptor (C structure may be even generated automatically).

Editing may work with descriptor only - it may easily contain metadata with defaults, limits etc. Documentation may be in same place

Upgrading configuration should be trivial in most cases - just encode JS object in new version. Helper functions may be added to handle complicated cases

Is this possible?

Is there some existing code that can be reused?

Can someone help me with this part?

Other ideas?

gaelj, 2016-07-18:
Will this need consideration regarding #2306 ?

ledvinap, 2016-07-18:
@gaelj : It will make most of MSP commands obsolete, so it will in fact partially solve MSP ID allocation.
And it will replace it with pgn allocation, but pgn space is larger (currently 16 bits, only 12 actually implemented), so it will be easier to manage (preallocate blocks for CF derived projects/implement allias pgn on configurator side/merge into main namespace eventually)

hydra, 2016-07-21:
@ledvinap there likely needs to be a PG list command which can tell a client which PG's are supported and at which version, and if they are an array the size of the array (total size or item count as you see fit).
and probably if a PG is per-profile too.

Given this information it should be possible to backup everything and restore compatible PGs and provide users with details on what setting could not be restored (if any).

Yes, some mechanism to describe each PG would be great, code generation would help with mismatches for sure.

hydra, 2016-07-21:
I now view this particular task as a critical task that will help us achieve the goal of removing the CLI.  See #922

hydra, 2016-09-12:
@ledvinap This is the only task that needs doing for the 1.14.0 which doesn't currently have an owner?  Can you take this on or shall I do it?

ledvinap, 2016-09-12:
Sorry, I'am a bit busy lately. I can implement CF part easily (it should be very easy), but Configurator part is more complicated, at least if elegant solutions is desired.
Maybe we can hard-code PG <->object mapping and improve it later ...

martinbudden, 2016-09-12:
Personally I'd leave this to the next release. It's not something you want to rush.

ledvinap, 2017-01-18:
(Looking at this a bit)
What do you think about https://github.com/jDataView/jBinary ?

My idea is to describe Parameter group in some platform-independent way. The description should contain enough information to (eventually) generate all related code (C struct, CLI binding, adjustment, display / generic display in configurator / documentation).

With jBinary, it should be quite easy to derive template from PG description and parse structures (both reading and writing). Ideally version information should be handled in this step - generating template for actual CF version (this should handle upgrade too - use different template when writing data back).

closed 2017-02-25 by hydra