cleanflight/cleanflight #2306: Clash of MSP values with betaflight

martinbudden, 2016-07-18:
The MSP values:
```
#define MSP_VOLTAGE_METERS              92 //out message         Voltage (per meter)
#define MSP_CURRENT_METERS              93 //out message         Amperage (per meter)
#define MSP_BATTERY_STATES              94 //out message         Connected/Disconnected, 
```
clash with the betaflight MSP values:
```
// Betaflight Additional Commands
#define MSP_PID_ADVANCED_CONFIG         90
#define MSP_SET_PID_ADVANCED_CONFIG     91
#define MSP_FILTER_CONFIG               92
#define MSP_SET_FILTER_CONFIG           93
#define MSP_ADVANCED_TUNING             94
#define MSP_SET_ADVANCED_TUNING         95
#define MSP_SENSOR_CONFIG               96
#define MSP_SET_SENSOR_CONFIG           97
#define MSP_SPECIAL_PARAMETERS          98 // Temporary betaflight parameters before cleanup and keep CF compatibility
#define MSP_SET_SPECIAL_PARAMETERS      99 // Temporary betaflight parameters before cleanup and keep CF compatibility
```
There may be other clashes, and there may also be clashes with iNav.

The immediate effect of this clash is that it makes it difficult to port some of the betaflight PID improvements to Cleanflight.

The longer term effect of this clash is that it is a compatibility break between the forks and means that there will be problems with third party configurators, OSDs and other tools.

I believe that `MSP_VOLTAGE_METERS`, `MSP_CURRENT_METERS` and `MSP_BATTERY_STATES` were recently added to Cleanflight, so I suggest a fix for this issue is that Cleanflight changes these values.

A for a longer term fix, I suggest we allocate a range of MSP values to be used by betaflight and another range to be used for iNav. This will allow compatibility of third party tools and also preserves the facility to port features between the three forks.

MSP values are a scarce resource, and need to be managed as such.

@hydra , @borisbstyle , @digitalentity  - your views?

borisbstyle, 2016-07-18:
All third party tools can do additional check for firmware identifier
too.....right?

martinbudden, 2016-07-18:
@borisbstyle , yes they can. But it doesn't make their job easy, and it also is a potential source of errors.

ledvinap, 2016-07-18:
Cleanflight repository may be used for MSP management - simply allocate new ID in https://github.com/cleanflight/cleanflight/blob/master/src/main/msp/msp_protocol.h, possibly commented out

@hydra?

gaelj, 2016-07-18:
One source of problems is when there is a delay in merging PRs containing new MSP values.
An author can issue a PR that uses new MSP values that are availbale at the time, but by the time it is merged, another fork has started using the same MSP value for something else.

digitalentity, 2016-07-18:
Best thing would be to allocate ranges of MSP IDs to
INAV/Betaflight/Raceflight and let the fork maintainers to manage them as
they see fit. 2-byte MSP IDs would really come in handy.

borisbstyle, 2016-07-18:
Best thing would be to have seperate MSP branch than, which is maintained
on same place and just reused as a library on different projects. Just like
any standardalisation works. There is no other way to make it 100%
bulletproof when multiple projects use it, which again also have multiple
branches and forks.

Multiwii also uses it. Not sure if that project is still alive?

But anyway I would rather love to see some kind of XML standard. Would be
much easier on different UI size too.

martinbudden, 2016-07-18:
Agreed that separating it off is the way ensure compatibility across forks. However, that will take a while to get into place.

In the meantime I think it would be good to get a short term fix in place, that is allocate some ranges for the different projects.

gaelj, 2016-07-18:
I checked in detail, those 3 values are the only conflicts between CF, BF and inav.

martinbudden, 2016-07-19:
@gaelj , in Cleanflight we have:
```
#define MSP_LED_STRIP_MODECOLOR         86 //out message         Get LED strip mode_color settings
#define MSP_SET_LED_STRIP_MODECOLOR     87 //in message          Set LED strip mode_color settings
```
whereas in betaflight we have:
```
#define MSP_LED_STRIP_MODECOLOR  127    //out message         Get LED strip mode_color settings
#define MSP_SET_LED_STRIP_MODECOLOR 221 //in  message         Set LED strip mode_color settings
```
which are the values you wish to mover forward with?

hydra, 2016-07-21:
yes, using CF codebase to allocate new ID's is fine.

Long term iNav and Betaflight should use the parameter groups for configuration storage and retreiveval.  Then only MSP commands for GET/SET/LIST PG's is required for all forks.  PG's have lots of available namespace and yes we can reserve ranges of IDs.

We should move to deprecate old MSP configuration commands that are replaced by one or more Parameter Groups, this will also free up MSP commands.

The blocker here is that Betaflight is still not using parameter groups.  @borisbstyle can you merge in the parameter group configuration changes into BF ASAP?  

@digitalentity I'm not sure if you are using them yet, can you do likewise.

The changes to support PG's are actually trivial even though there are quite a lot of places that need changing.  Long term this will also provide a much needed backup/restore process and the removal of the dreaded CLI so it's in our interests to get this done ASAP.

hydra, 2016-07-21:
@ledvinap did you implement PG GET/SET via MSP?  I think you said you wanted to do this?

martinbudden, 2016-07-22:
@hydra , 

Parameter groups are targeted for betaflight 3.1 see https://github.com/betaflight/betaflight/issues/624 . Release of 3.0 is fairly imminent.

Parameter groups are targeted for iNav 1.3, see https://github.com/iNavFlight/inav/pull/151 . Release 1.2 of iNav is fairly imminent.

On a semi-related issue I think it is important for Cleanflight to move over to the new IO. This will allow convergence of the device driver code, and possibly also the sensor and IO code.

closed 2017-02-25 by hydra