cleanflight/cleanflight #2315: Flight- Bad Drift issue in Horizon/Angle modes

rgluckman, 2016-07-20:
3 of my Quads drift heavily in Angle/Horizon modes. 
All are using Cyclones. 

MotoLab Cyclone FC
BLHeli-14.6
Betaflight 2.9

No matter how many times I calibrate the Accelerometer and do an "in flight calibration" my quads won't hold a level hover. The more I fly the worse it gets. It pulls in all directions. 
I tried silicone dampeners on the FC thinking it was a vibration issue. That didn't work either. 
My TBS Powercube and Naze32 Rev5 hold Angles modes just fine. 
What else can I do to fix this?

andre92, 2016-07-20:
Looks like an issue with Betaflight 2.9. Please post it here: https://github.com/betaflight/betaflight/issues
Does this also happen with Cleanflight 1.13 with same settings? (like gyro_lpf, gyro_soft_lpf, and acc_soft_lpf) Or in older Betaflight versions?
Betaflight 2.9 seperated attitude update from pid loop to only 100 times per second. Cleanflight does it every loop.

rgluckman, 2016-07-20:
Thanks Andre! I'm losing sleep trying to figure this out. 
I will post in the Betaflight section right now. 
I will try Cleanflight 1.13 tomorrow and test that out.

hydra, 2016-08-10:
closing, no activity. use betaflight forums for betaflight support please.

closed 2016-08-10 by hydra