cleanflight/cleanflight #2316: PID controllers

xdigyx, 2016-07-20:
I have just tested LuxFloat controller from FW 1.13 on CC3D and personally I believe changing LEVEL D term from value 20 to 100 doesn't give visible changes in my copter bahaviour. Oposite to this for  I term pitch/roll/level smallest value change gives big difference. This is probably a result of scaling made for this FW release.

andre92, 2016-07-20:
Lux got new scaling in 1.13. It now scales the same as Rewrite so you can change controllers more easily. You may need to re tune if you were using lux in 1.12.

xdigyx, 2016-07-20:
I istalled clean FW and set LuxFloat straight away, so I should have defaults. In my opinion instead of scaling them inside FW, it would be much better to do the scaling within the config tool, so once you change the controllers, the config tool either should recalculate or set defaults for the new PID controller. The defaults which were set while changing from defaults pid controller to LuxFloat are not usable.

xdigyx, 2016-07-23:
I have switched to MWRewrite and had to retune again fortunetelly not much.
Another settings trouble I found with is baro usage: chanigng alt PIDs actually does not give much difference - from P 0,1/I 0,001, D 1 till P 10/I 0,050, D 100. The model is rapiddly changing its atl. I wanted to make it smooth, but PIDs changes give nothing.

hydra, 2016-08-23:
this issue seems to be a collection random statements/opinion and questions rather than something we can work on.  the subject is also very misleading, i don't know what to do other than close this issue as invalid.

closed 2016-08-23 by hydra