cleanflight/cleanflight #2318: Possible Support for RGBW LED strips?

adamweld, 2016-07-20:
Hi, I've used the WS2811 based LED controls in Cleanflight before and they work great, awesome feature. I recently picked up some new strips of WS2812 LEDs that are RGBW from Adafruit. Here's the datasheet:
https://cdn-shop.adafruit.com/product-files/2757/p2757_SK6812RGBW_REV01.pdf

The only real difference is that they have a White LED as well as the RGB LEDs. I was wondering how hard it would be to make Cleanflight support these LED strips, as they can generate nicer colors than their RGB bretheren.

Thanks in advance for any response!

gaelj, 2016-07-21:
The difference is in the data transmission protocol.
Packets size grows from 24 to 32 bits ; the 8 extra bits store the W value.
LED type cannot be detected so it would have to be a configurable option.
By the way WS2812 are RGB, not W. You are referering to SK6812RGBW. WS2811 is the standalone chip, WS2812 is integrated in the LED module.
A combination of S and V of the HSV can be used to calculate the W value.
A few driver files (light_WS2811....) will need updating.
The increase in DMA buffer size will impact RAM footprint by 32 bytes.
I've prepared a PR, could you try it out ? I can only test on RGB LEDs here.
Just compile & flash, restore settings, then in cli enter `set led_type = RGBW` then `save`.

gaelj, 2016-07-21:
Let me know what board you use if you need a compiled hex.

adamweld, 2016-07-22:
Sure, thank you. I'm using the RMRC Dodo board.

gaelj, 2016-07-22:
[cleanflight_RMDO.zip](https://github.com/cleanflight/cleanflight/files/377240/cleanflight_RMDO.zip)

gaelj, 2016-07-23:
@adamweld did you test this ?

gaelj, 2016-07-24:
@adamweld here is an updated hex file, containing a bugfix (CLI feature was only available if blackbox was enabled) + cleanups + merge with latest master (meaning new LED strip code, requiring Configurator update).
The new configurator enables you to edit a color by double-clicking it (3 sliders to control HSV values will appear). So you can play with the S & V parameters and report back how it works out.
[cleanflight_RMDO.zip](https://github.com/cleanflight/cleanflight/files/380038/cleanflight_RMDO.zip)

adamweld, 2016-07-24:
Actually, the quad with the RGBW LEDs was not using a Dodo, but a NAZE target. I built your PR myself with for it last night and flashed it, but for some reason it's not connecting to the configurator. I need to take it apart a bit to get to the boot pads but once I've done that I will try again with your updates and see how it goes.

adamweld, 2016-07-24:
If you can provide a NAZE hex that will make things even easier.

gaelj, 2016-07-24:
Sure.
[cleanflight_NAZE.zip](https://github.com/cleanflight/cleanflight/files/380194/cleanflight_NAZE.zip)

gaelj, 2016-07-24:
This one has the last commit where I put back MSP_LED_STRIP_MODECOLOR id's to the old values (I hadn't realized the configurator had been reverted too).
[cleanflight_NAZE.zip](https://github.com/cleanflight/cleanflight/files/380472/cleanflight_NAZE.zip)

adamweld, 2016-08-01:
Hi gaelj - I was finally able to flash your version onto my Naze, after struggling a bit to get the board into bootloader because my BOOT pads were somehow ripped out by the previous owner. Eventually I had to solder to the pins of the STM32 but I got it to flash ;)

So, the LEDs start up now and keep a consistent color - which means that your change seems to have worked. However when I try to access the LEDs page of the configurator, I lose connection with the board (packets start dropping and I must reconnect).

gaelj, 2016-08-01:
OK sorry, the configurator's been updated.

adamweld, 2016-08-01:
With new configurator and last hex, I can open the LED tab but the settings don't seem to have any effect. LEDs stay lit with a seemingly random set of colors.

When updated to the newest HEX, the LEDs are now staying off. I'm going to try the previous version again to make sure its a repeatable problem.

gaelj, 2016-08-02:
I have an RGBW ledstrip on it's way any day now, so no worries.
In the configurator set your LEDs to Color mode, then you can edit the HSV values of the selected colour by double clicking it. Try playing with S and V, this should alter the white LEDs brightness.

adamweld, 2016-08-02:
Will do. Thanks so much for the dev work & sorry about my slow responses...

r1b4z01d, 2016-08-30:
@gaelj I can help test this if you need.

gaelj, 2016-08-30:
Thanks, I have received one from eBay myself a few days ago. It's now
tested, debugged and working at my desk, I'm now thinking about how to
benefit from these white lights in the preset LED functions and adapting
the configurator. I'll update the PR with latest changes tonight so you can
have a try if you want.
Any ideas regarding how to benefit from the new White Color are welcome.
I was thinking about blinking white for example (in addition to the
existing blinking to black)
I also modified the Larson Scanner.

gaelj, 2016-08-30:
FYI testing shows the spec is wrong regarding color ordering in the frames sent to the LED controlor.
Instead of 
```
    uint32_t rgbw = (color.rgbw.r << 24) | (color.rgbw.g << 16) | (color.rgbw.b << 8) | (color.rgbw.w);
```
It should be
```
    uint32_t grbw = (color.rgbw.g << 24) | (color.rgbw.r << 16) | (color.rgbw.b << 8) | (color.rgbw.w);
```

adamweld, 2016-08-30:
Flash to white sounds good. I can also think of an arbitrary "white brightness setting" besides color settings.

gaelj, 2016-08-30:
Setting the W value for predefined colors is already possible in CLI (I've added a 4th parameter to the `color` command for this).
The CFC update will also have the corresponding slider added to the color customization box in the GUI.
For the moment the only default color that uses the W LED is white.

Red13x3, 2016-08-31:
Can I make an extension to this suggestion here or do I make a new "Issue"?

The idea is to link an AUX channel to an LED strip(s).

Reason: To be able to set the colour of the LEDs on the fly, to match the VTX frequency you are allocated in a race, and thus make it easier for people to follow where racers are.

This is probably big enough for a new issue... But wanted to bring your attention to it as there are some clever coders in this thread that know what they are doing with LEDs. :)

gaelj, 2016-08-31:
I did that for betaflight already. I can easily port it to CF in a separate
PR.

Red13x3, 2016-08-31:
@gaelj  Yes, please!! :)

ledvinap, 2016-08-31:
It may be better to use adjustments for this ... (or #2378)

Zethyl, 2016-11-23:
Issues with LED Strip on OMNIBUS F3 Pro target OMNIBUS 3.0.1

board has a dedicated LED Strip connector

15 LEDs total. #0-8 are a strip connected to the strip of #9-#14 by thin servo wire

SK6812 RGBW LEDs

-no control of led #12+, always white
-random control of #0-11, if i set all to red i get R,G,B,R,G,B,R,G,B,R,G,B,W,W,W if i set to white i get all white, if set to blue i get B,R,G,B,R,G,B,R,G,B,R,G,W,W,W

closed 2017-02-25 by hydra