cleanflight/cleanflight #2319: GPS reception is inherently unstable in 1.13.0

jflyper, 2016-07-21:
I'm looking at an unstable behavior of my GPS for half a day or so, and found out that the GPS reception in 1.13.0 is inherently unstable: it configures SVINFO messages to be sent, which in some cases causes serial receive buffer overflow, dropping lots of bytes and eventually dropping lots of messages.

Here's an example of what's happening:
- Let's say we are in UBX mode at 38400bps.
- With my Saleae in my house, I can see that there are tiny, short, and long burst of chars sent; long ones last for 130msec.
- At 3840 byte/sec, this is around 500 bytes.
- TASK_GPS runs at 10Hz, which is at 100msec interval.[https://github.com/cleanflight/cleanflight/blob/v1.13.0/src/main/scheduler_tasks.c](url):L99
- Serial receive buffer is 256 bytes. [https://github.com/cleanflight/cleanflight/blob/v1.13.0/src/main/drivers/serial_uart.h](url):L26-35
- Suppose a long burst begins immediately after the GPS task was ran.
- The burst will begin to fill the serial receive buffer.
- 66msec later, the 256 byte buffer is full and begin to drop incoming chars.
- The buffer full condition lasts for 34msec until the GPS task is ran again. During this period, 130 chars are dropped.
- When the GPS task is ran, the receive buffer becomes empty, storing remaining 114 chars.

For me, a pragmatic fix was to make the GPS task run at twice the frequency, or disabling the SVINFO completely (effectively disabling the signal strength bar graph). The task frequency method is vulnerable, since the size of the SVINFO message may grow larger depending on a number of visible sats, possibly causing another task throughput and/or buffer overflow problem. So, as long as we want the bar graph feature, the vulnerability can not be resolved unless there's some way to control (limit) the number of sats in SVINFO messages.

I haven't tested this yet, but lowering baud rate may help, since it effectively spreads incoming chars evenly throughout the bandwidth. The basic rule is:

         baudrate / 10 * task_interval < receive_buffer_size

This may be the reason that GPS on soft serials at low speed are reportedly work (19200/10 * 0.1 = 192 < 256). But of course, how or can ublox fit configured messages into this bandwidth is another problem.

Modifying the GPS code to interrupt driven may liberate us from the rule, but I'm not sure if it will cause yet another problem in the upper layer or not.

jflyper, 2016-07-21:
iNav 1.1 schedules at 25Hz (40msec interval), which is just about right for 57600bps.
[https://github.com/iNavFlight/inav/blob/1.1/src/main/scheduler/scheduler_tasks.c#L88](url)

        57600/10*0.04 = 230.4 < 256

ledvinap, 2016-07-21:
@jflyper : IMO it would be best to increase TASK_GPS frequency so that it can handle all possibly received  data at maximum speed, or even higher to spread data processing into shorted chunks.

Using 100Hz at 115200 will result in chunks of 115 bytes, which seems reasonable.

GPS task handles all received bytes and calls packet processing only when full packet is received, so increasing GPS update frequency will cost only slightly more than scheduler overhead. Processing GPS data in smaller chunks will reduce maximum task processing time, improving scheduler performance.

Also running GPS task at same frequency as GPS update rate will cause worst possible jitter (GPS update is delayed between 0  - 100ms, latency will be modulated by difference between GPS_TASK frequency and actual GPS update rate)

IMO we should set TASK_GPS to 100Hz now. When scheduler is improved, GPS task may be triggered by data waiting is serial buffer.

jflyper, 2016-07-21:
@ledvinap 
+1

100Hz makes sense.

hydra, 2016-07-21:
@jflyper excellent investigation work.  I have also been looking into why the SVINFO packets were being lost off and on for a while now which is when I found the bug in the UBLOX parsing code, it's excellent to see someone else having a look into this too.  Will merge fix now.

hydra, 2016-07-21:
fix merged.

closed 2016-07-21 by hydra