cleanflight/cleanflight #2326: The motolab target doesn't work with SPI black box chip tie it to the Same SPI2 as the tartget call out.

bnn1044, 2016-07-25:
The motolab target doesn't work with SPI black box chip tie it to the Same SPI2 as the tartget call out.

I make a prototye for the motolab SPI target with the Flash Chip tie to the same SPI2.   the Gyro CS pin work when the CS is high. when Pulling the CS pin low. it should select the blackbox flash Chip. on disarm. on the GUI, it see the black box chip. but as soon as I enable the Blackbox feature and arm the quad. the gyro data go crazy. like infinite roll and pitch.  Can you please look at that? thanks,

MJ666, 2016-07-25:
I think it is not a good plan to use the same SPI port for the MPU and the SPI flash. You should use two independent ports. For CS there should be also two independent pins used.

bnn1044, 2016-07-26:
Does it work? then why that pin out stay in the target file in the MOTOLAB target if is not working?
Thanks .
I know If I use difference SPI it will work. just wondering if the code can take care of same SPI two devices?

jflyper, 2016-07-26:
@bnn1044 What happens if you reduce the blackbox bandwidth to, say, 1/32 or even 1/100?

ledvinap, 2016-07-26:
@bnn1044: Both SPI and GYRO chipselect ping are active-low.
The code seems that it may work. Using separate CS pin for each device is better option. Theoretically it may work with inverter before SPI FLASH chipselect with modified code, but IMO it if not worth the trouble

MJ666, 2016-07-26:
The MPU need full speed. It is not good practice to share an SPI port with oter funtions. The SDCard is also using an different initialisation then the sensors for the SPI port.

ledvinap, 2016-07-26:
@MJ666 : Both accgyro and blackbox are using blocking access to attached devices. It does not matter if bus is shared or not ..

jflyper, 2016-07-26:
Oh, I didn't notice/understand that the flash and the gyro are sharing a same CS pin.
> Theoretically it may work with inverter before SPI FLASH chipselect with modified code

Yes,  and the mod is not only limited to CS pin programming, but should include switching between CPOL=1,CPHA=1 for gyro and CPOL=0,CPHA=0 for flash.

> The MPU need full speed

It's off topic, but out of pure curiosity, I wonder how jitters on gyro read intervals affect flight behaviors. After all, since gyro is only using a fraction of the available bandwidth, it's interesting to see if we can fit other transactions onto the same SPI bus.

bnn1044, 2016-07-26:
@MJ666 : Both accgyro and blackbox are using blocking access to attached devices. It does not matter if bus is shared or not ..

Only problem may be id MPU and SPI flash are using different speeds, but used MPU registers are IMO in 20MHz space(using 9MHz bus speed), so using SPI flash speed (18MHz) should be fine ...

So set the Gyro At 8Kz. and what flash Speed should I set it too? To match the speed of the gyro?

ledvinap, 2016-07-26:
@jflyper : CPOL/CPHA is different for SDcard, not SPI flash (if I read the code correctly)

Gyro read is synchronized almost perfectly to data ready - gyro is scheduled a bit earlier and busy-waits for new data. But IMO it is overkill ...

bnn1044, 2016-07-26:
@jflyper : CPOL/CPHA is different for SDcard, not SPI flash (if I read the code correctly)

Gyro read is synchronized almost perfectly to data ready - gyro is scheduled a bit earlier and busy-waits for new data. But IMO it is overkill ...

I am not using the SDcard, I am using the flash chip.

if it is impossible to do the same SPI chip with SPI gyro. later on if we have more device. like OSD chip using the SPI too. that will be limited to our hardware design.

jflyper, 2016-07-26:
Ah, I was assuming the inverter for CS switching, and thinking of a scenario that CS switches from gyro to flash too fast, before SCLK goes high (idle) again, then the flash might see idle SCLK low. But this scenario doesn't happen (I hope).

jflyper, 2016-07-26:
[FYI]
Arduino `SPI` library supports the `SPISettings` object that master can specify for a transaction. The `SPISettings` includes clock, bit order and mode.

closed 2016-07-28 by bnn1044