cleanflight/cleanflight #2330: FailSafe Mode

mrduncan626, 2016-07-26:
I'm currently using CF 1.13.0 with the mixer set to a Custom Tricopter.  Flight controller is a Flip32.  FailSafe Procedure is set to Land.  When I ground test the failsafe function, it goes into a drop failsafe rather than a land failsafe mode.  Pic below indicates my failsafe setup.  Is there an issue in 1.13.0 associated with the Land Failsafe Procedure?

FYI, it was working in 1.12.0.

[image]

andre92, 2016-07-26:
What failsafe have you set for your receiver? Preset or no pulses?
Have you tried setting throttle to hold? Especially if you use motor_stop. Maybe the FC doesn't want to restart idle motors.

xdigyx, 2016-08-01:
got same problem with NAZE32 rev6. My hexa hovered 2m over the ground. The situation is quite strange. although conf-tool sees THR at 960-970, Bbox logs show values 1150-1180 and rxsignalreceived is all the time 1. THR value of 2nd stage i have set to 1350, but motors go directly to 1000, so it seems that the FC disarms straight away.
t

xdigyx, 2016-08-01:
there is one thing to change: Minimum Length to 980-990 and then it works...

mrduncan626, 2016-08-02:
Negative on setting the FailSafe Minimum Length to 990.  I have several F1 FC's which are responding the same.  Tried installing the latest BetaFlight also but the FailSafe doesn't function properly either.  However, on my F3 FC's, the failsafe works as advertised but I have not installed CF on my F3 FC's.  It's almost soundly like a F1 FC issue.

hydra, 2016-08-10:
The failsafe is the same between F1/F3 (IIRC).

Some logs and config dumps required to progress this further.

Exact steps to reproduce also required please.

mrduncan626, 2016-08-14:
Which over to BetaFlight and the failsafe on the F1's have the same problem.  I have 5 - F3 quads and the failsafe works as advertised.  I have 2 - F1 quads which are having the same failsafe issue.  ALL failsafes are configured the same.  This same F1 failsafe issue seem to be present in both CF and BF.

Attach is a BF dump if you can use this to determine what's going on.  If not, I'll have to reinstall CF.

[F1 CLI Dump.txt](https://github.com/cleanflight/cleanflight/files/417084/F1.CLI.Dump.txt)

xdigyx, 2016-08-16:
Regarding Fail safe on 1.13, I actually have a different issue:
form time to time my model freezes and I am not sure whether I am loosing RC link or something gets wrong with FC. So I wanted to use blackbox to find this out. So firstly I wanted to see what would be visible and did quick test on a bench. But actually it seems to be difficult to catch it. Why? bcause even when I turned off the RC TX, the variable rxSignalreceived is all the time "1" and failsafePhase is "0" and rcCommand[THR] does not set to the real value [on my RC RX FS value  is set to 980, FC FS value 995). The FS actually seems to work as it changes THR to the requested value, but under this circumstances I can not be sure what sould I expect during a freeze my model catches...

Pleasae verify/correct the FW.

hydra, 2016-08-27:
i will need a CF dump please.  Also exact steps to repeat would be very useful to speed things up, provide as much info as you can.

rb1205, 2016-08-29:
Is the throttle value correct? 1150 is barely the idle throttle value for most configurations.

closed 2017-02-25 by hydra