cleanflight/cleanflight #2333: Eclipse + GDB openOCD debuging fail : I can't access FC by SWD, but I can burn the hex file.

qwedsazxc78, 2016-07-28:
## Hi guys, I want to develop the our compass into SPRacingF3, but I meet below problem.

**problem:**
- Eclipse openOCD can't connect the flight control board by SWD correctly, the window message show 
- I can burn the hex file by SWD, STLINK utlity, and USB burnner, so I believe the HW is ok. all setting picture in goolge driver: https://goo.gl/awIz7E
- I think I have to configure something to let the OpenOCD can connect the stm32f303CC MCU. Therefore, I try to modify the reset_config option of OpenOCD cfg file as below

> 1. reset_config srst_nogate 
> 2. reset_config srst_only
> 3. reset_config srst_only srst_nogate
> 4. reset_config none

I try it about 2 week, so I need some nice boy help me fix it.
Also, all I follow this tutorial to configure my environment. 
https://github.com/CraigJPerry/cleanflight/blob/dev-docs/docs/development/Building%20in%20Eclipse.md#install-eclipse

**environment**

- win10
- eclipse-cpp-luna-SR2-win32-x86_64
- gcc-arm-none-eabi-4_9-2015q3-20150921-win32
- gnuarmeclipse-build-tools-win64-2.6-201507152002-setup
- gnuarmeclipse-openocd-win64-0.10.0-201601101000-dev-setup
- gnuarmeclipse-qemu-win64-2.4.50-201510290935-dev-setup
- STM32 ST-LINK Utility v3.9.0
- STM32F4 USB dongle + SPRacing F3 10DOF board

qwedsazxc78, 2016-07-31:
hello guys,
I resolve this issue, and the solution is that I change the setting back to default, so I find the problem is from below two diagram.

The context of run and reset commend should be empty, and in the debugger setting, I use the stm32f3discovery setting. then, everything go back to perfect status. 
[image]
[image]

SteveAmor, 2016-08-01:
Good to hear you resolved the problem. Can you please close this issue.

qwedsazxc78, 2016-08-01:
I try the below method and fix the problem. close issue

closed 2016-08-01 by qwedsazxc78