cleanflight/cleanflight #2339: Newly Built Quad w/ BIGGIE ISSUE

Raptor1812, 2016-08-03:
Hi Guys,

I just finished building an AstroX mini quad. I took it out today for the first time and notice when I do some type of acrobatic (flip/rolls/yaw spins) move it some times goes and does some crazy type of out of control spin flipping. Like it has a mind of it's own. It also doesn't happen every single time but more times than not. If anyone could give me any experience or suggestions on what this could be I'd greatly appreciate it. 

Basically stock PIDs

     P      I       D
R  4   .040    19
P  4   .040    19
Y  9   .045     0

Blackbox 8/2/16 log:
https://www.dropbox.com/s/ww48j9utp2hbt4r/blackbox_log_2016-08-02_ISSUE.TXT?dl=0

 Video:
https://www.dropbox.com/s/k75n3dcsjpzh002/QuadIssue.MP4?dl=0

[blackbox_log_2016-08-02_ISSUE.TXT](https://github.com/cleanflight/cleanflight/files/398501/blackbox_log_2016-08-02_ISSUE.TXT)

*PLEASE SOMEONE HELP ME!

hydra, 2016-08-10:
please use rcgroups.com for support.  If further investigation reveals a bug please raise it here but the first line of support is forums and irc.

hydra, 2016-08-10:
futhermore, using CAPS and non-factual language in titles etc to get attention will certainly not help, don't do it please.

closed 2016-08-10 by hydra