cleanflight/cleanflight #2343: About calculating error in the system

Quicktimer, 2016-08-05:
Hello there,

I have one question about that how the system calculate an error. A few day I read the code in github but I do not understand. 
System output is gyroADC/4
rcChanel is Pitch/Roll/Yaw/Throttle - for all except throttle range is +-500

How error is calculated? error = +-500 - gyroADC/4. Is it true?

Thanks and best wishes!

Quicktimer, 2016-08-05:
Used PID MW naze32 v1.12

andre92, 2016-08-07:
That is legacy from old Multiwii code. Multiwii used to divide Gyro ADC by 4 in the IMU to lose some of the noise. Cleanflight is using the exact same PID controller as MultiWii so in order for it to work the same, the Gyro ADC gets divided by 4.

closed 2017-02-25 by hydra