cleanflight/cleanflight #2344: Feature request: attitude in place of acc values in FrSky S.port telemetry

jelinj8, 2016-08-05:
Hello,
it would be nice to be able to send angles back to Taranis (e.g. for artificial horizon telemetry screen).
I have recompiled Cleanflight with this, but that was quite an old version.

Instead of
```
smartPortSendPackage(id, accSmooth[X] / 44);
smartPortSendPackage(id, accSmooth[Y] / 44);
```
I had
```
smartPortSendPackage(id, attitude.values.roll / 10);
smartPortSendPackage(id, attitude.values.pitch / 10);
```

I'm not sure if it is a correct way of doing this.
Alternative would be to add custom sensor ID and force Taranis to query that.

ledvinap, 2016-08-06:
Taranis will poll all 32 sensor adresses (3 bits of adress are checksum).
With openTx 2.1 is is possible to define your own sensors (mapping of telemetry ID to telemetry value), so fot personal use just use some telemetry IDs.
For public use it would be best to get telemetry IDs 'officially', but I don't know if some mechanism is in place ...

hydra, 2016-08-10:
agreed.  @jelinj8 can you contact the opentx guys and get some IDs for this?  I feel it could be useful too, and is quite common data among things that fly :smile:

jelinj8, 2016-08-10:
OK, I'll try to contact them, but I'm afraid that there is no assignment
mechanism, there are implemented just IDs of existing genuine FrSky
sensors.

jelinj8, 2016-08-16:
I have created a ticket for this in OpenTX github ( https://github.com/opentx/opentx/issues/3700 ).

jelinj8, 2017-01-14:
It seems that we are on our own with this - OpenTX issue closed.

closed 2017-02-25 by hydra