cleanflight/cleanflight #2347: 3d_deadband_throttle gets incorrect value when saving RX config in GUI

andre92, 2016-08-09:
After changing Receiver settings in the GUI (things as RC deadband, expo, throttle expo), the 3d_deadband_throttle setting in CLI is 30212. With this setting, arm switch will immediately start motors when using 3d mode (should only start at mid stick)

KiteAnton, 2016-08-09:
This seems very similar to the same issue in betaflight, it was fixed with [this PR](https://github.com/betaflight/betaflight/pull/929). And then [this PR ](https://github.com/betaflight/betaflight-configurator/pull/194)for the configurator.

hydra, 2016-08-10:
Fixed in latest development configurator.  Please test and report any issues in configurator firmware.

3D Throttle Deadband setting moved to receiver page where it belongs.  3D ESC settings remain on configuration page and the '3D' title has been updated to '3D ESC' to clarify.

closed 2016-08-10 by hydra