cleanflight/cleanflight #2348: Convert Cleanflight to use new make system as used by betaflight and iNav

martinbudden, 2016-08-10:
New make system has several advantages:

1. Removes much of the target-specific code from the codebase
2. Builds more quickly
3. Makes it much easer to add new targets

@hydra , one enhancement I'd like to make to the new make system is to use the gold linker - linking takes a significant part of the build time (I'd guess about half), and so speeding up linking would significantly speed up the over build. Do you have any experience with the gold linker, or could you look into it?

Fast build times are important for at least two reasons:
1. Removes the disincentive to doing `make all` before a PR submission
2. Speeds up travis continuous integration (which can be a bottleneck)

jflyper, 2016-08-11:
If Makefile is going to be tweaked, why not suppress linker command line echo?

It prevents shell-make-ers from catching stupid errors.

martinbudden, 2016-08-11:
Closing, since this is a duplicate of issue https://github.com/cleanflight/cleanflight/issues/2274 which I'd forgotten about.

closed 2016-08-11 by martinbudden