cleanflight/cleanflight #2349: Quick question: SPI SCK pullup for sdcards: source?

jflyper, 2016-08-11:
Quick question regarding SPI initialization.

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/bus_spi.c#L90-L93

The comment says "In clock-low mode, STM32 manual says we should enable a pulldown to match". Can some one give me a pointer to the part of the manual?

hydra, 2016-08-11:
@jflyper No I can't find it.  I wonder if @thenickdude knows where it is?

thenickdude, 2016-08-11:
It's a note in the "clock phase and polarity controls" section of the "RM0316" reference manual:

>The idle state of SCK must correspond to the polarity selected in the SPIx_CR1 register (by pulling up SCK if CPOL=1 or pulling down SCK if CPOL=0).

closed 2016-08-11 by thenickdude

jflyper, 2016-08-11:
@thenickdude @hydra Thanks!!!

But then I started to wonder why non-sdcard case (CPOL=1) don't set pull-up...

thenickdude, 2016-08-11:
It probably should to be honest, I only wrote the SDCard case :). I suspect
the requirement for the pull-up/pull-down only becomes important if the SPI
block is ever disabled, since it'll probably stop actively driving the
clock as an output.

thenickdude, 2016-08-11:
Ah I guess it doesn't set any pull-up/pull-down because the clock polarity
isn't decided at this point, it depends on what device will be used on the
SPI bus later (some accelerometer or a flash chip, which could have
different polarity preferences).

jflyper, 2016-08-11:
That's what I thought initially, but looks like drivers for SPI devices use the settings as is; they only manipulate the clock divisor.

Well, MPUs has been successfully accessed trillion times by now, so it works pretty damn good without pull-ups. I'll probably do some further readings and logic-scoping on both cases.

I appreciate your help. Thanks again!