cleanflight/cleanflight #2350: naze32+Blheli xm20a+Naze32 rev5 = No sync

RedStider70, 2016-08-11:
Hey guy's, i have a problem with my Blheli xm20A v1 esc's because, when i calibrate them in cleanflight,and i test them with master slider, are perfect.. But, when i use my radio (Taranis), every motor spin up slowly or faster then other.. Do you have a solutions for this problem? The RX is a D4R-II.

Arakon, 2016-08-12:
There is no solution as that is perfectly normal. The copter is vibrating and not perfectly level, so the PID controller is attempting to level it out, causing different motor speeds.

RedStider70, 2016-08-12:
Ok..but for solve this problem, i need to isolate vibration to naze or other?

Arakon, 2016-08-12:
There's no "solving" this as this is NOT a problem at all. That's simply the software doing its job. If they all ran at the exact same speed, the copter wouldn't fly except for straight up.

closed 2016-08-12 by RedStider70