cleanflight/cleanflight #2353: rxSignalreceived & failsafePhase variables possible bug

xdigyx, 2016-08-17:
Regarding Fail safe on 1.13 (CC3D) or actually variables visible in BlackBox:
form time to time my model freezes and I am not sure whether I am loosing RC link or something gets wrong with FC (this is less important thing). So I wanted to use blackbox to find this out. So firstly I wanted to see what would be visible and did quick test on a bench. But actually it seems to be difficult to catch it. Why? bcause even when I turned off the RC TX, the variable rxSignalreceived is all the time "1" and failsafePhase is "0" and rcCommand[THR] (and this is more important) does not set to the real value [on my RC RX FS value  is set to 980, FC FS value 995). The FS actually seems to work as it changes THR to the requested value , but under this circumstances I can not be sure what sould I expect during a freeze my model catches...

Pleasae verify.

Log attached.
[test.TXT](https://github.com/cleanflight/cleanflight/files/420665/test.TXT)

hydra, 2016-08-23:
Sounds likely that you're loosing RX for a short period of time and that failsafe is working OK.

For us to investigate any further you'd need to provide details on how to repeat your problem, include a config dump and provide details of the TX/RX you're using.

To me this seems like more of a support request, mis-understanding of how the failsafe system works or mis-configuration of the TX/RX/FC.  Without more information we can't help further.  Further help should be asked via support channels first and then, if you can identify this as a bug, please provide more information and re-open the issue here.

closed 2016-08-23 by hydra

xdigyx, 2016-08-23:
hydra,
In terms of FS, it works fine.
And most probably the problem is with my RX.
What does not seem to work is the rxSignalreceived and failsafePhase, as these do not change at all:
rxSignalreceived is always 1, even if I turn off my RC TX (and RX PWM is lower than time for FS set in CL).
As well as failsafePhase, which is always 0, even when it's clearly visible than 2nd phase is actually activated.