cleanflight/cleanflight #2354: IO SP Racing F3 Evo

urlu75, 2016-08-17:
Hi,

Just got an official SP Racing F3 Evo flight controller and contribute to this great project.
I will use the board with : 
- OSD (UART1)
- GPS (UART2)
- SBUS RX (UART3)

but where to plug Smart Port telemetry ? What are the plan for SoftSerial with this board ?

thanks,
lulu

andre92, 2016-08-18:
I have same issue with this board. If not all 8 motor outputs are on one timer, maybe some motor pins could be used for it? At least with quadcopters.

gaelj, 2016-08-21:
Try this https://github.com/cleanflight/cleanflight/pull/2362

hydra, 2016-08-23:
@urlu75 Please provide feedback on #2362 when you have tried it.

I will close this so we can keep discussion in on the PR.

closed 2016-08-23 by hydra