cleanflight/cleanflight #2359: prerequisites to arm via MSP USB port

Abu-Abdullah, 2016-08-19:
Hi, 

i would appreciate some help to identify what are the prerequisites to arm via MSP USB port.
I'm writing a java code to control the FC from my PC but not able to arm using via MSP_SET_RAW_RC
it is working through the configurator but not able to do it through usb port.

MSP_SET_MOTOR is working for me so i think the main code to format the commands is working. the code is taken from flone-android project but im using normal java not android code

is there any configuration i need to set before sending MSP_SET_RAW_RC commands. 

```
# dump

# version
# Cleanflight/SPRACINGF3EVO 1.13.0 May 30 2016 / 22:43:16 (0245144)
# dump master

# mixer
mixer OCTOFLATX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature VBAT
feature FAILSAFE
feature TELEMETRY
feature CURRENT_METER
feature RX_MSP
feature RSSI_ADC
feature BLACKBOX
feature TRANSPONDER

# map
map AETR1234

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
[... 410 lines cut ...]
			if (comPort[i].getSystemPortName().equals("COM4")) {
				comPort[i].setBaudRate(115200);
				comPort[i].setParity(SerialPort.NO_PARITY);
				System.out.println("COM4 open? "+comPort[i].openPort());
				comPort[i].setComPortTimeouts(SerialPort.TIMEOUT_READ_BLOCKING | SerialPort.TIMEOUT_WRITE_BLOCKING, 10, 0);

				int h=0;
				do {
					h++;
					// Order: Roll Pitch Yaw Throttle AUX1 AUX2 AUX3 AUX4
					byte b[] = SendRequestMSP_SET_RAW_RC(new int[]{1500, 1500, 2000, 1000, 2000, 1000, 1000, 1000});
					comPort[i].writeBytes(b, b.length);
					Thread.sleep(20);
				}while(h<100);

				comPort[i].closePort();
			}
		}
	}

	public static void main(String args[]) throws Exception {
		new Test();
	}
}
```

hydra, 2016-08-23:
to answer your question, no prerequisites to use SET_RAW_RC.   I haven't time to go over your code to find the solution.  Perhaps connect up a USB-UART to a spare serial port and connect configurator to USB and send MSP commands to spare serial port so you can see if your commands are being received and applied - e.g.  monitor the receiver tab while sending commands via another serial port.

hydra, 2016-08-23:
also try 'map AERT1234' to match the RX_MSP / SET_RAW_RC behavior.

closed 2016-08-23 by hydra

Abu-Abdullah, 2016-08-24:
thanks, that was useful, it was mapping issue as you mentioned