cleanflight/cleanflight #2360: In-flight switchable max_angle_inclination

danarrib, 2016-08-20:
Hello guys!
It would be nice if we could set an AUX channel switch to adjust the **max_angle_inclination** value in flight. That will allow pilot to make the quadcopter faster or slower in ANGLE mode by just a change on a switch on TX.

Today the max_angle_inclination variable is GLOBAL. So if it's changed to be a PROFILE variable, we can switch profiles easily.

hydra, 2016-08-23:
profiles do not switch in-flight.  moving it into a rate profile would make more sense as they *are* switchable in-flight.

rb1205, 2016-08-29:
IMHO it's simpler and quicker to just set up the maximum angle you'll use and then limit the range on the radio side. I agree on to the switch to the profile variable thought.

danarrib, 2016-08-29:
The problem is sometime we want a "slow flying" behaviour to allow people new to hobby to get the quad for the first time, and then make the quad little faster when they get some skill.

I usually fly rate mode, but when somebody want to fly my quad, I put it in the Angle mode with 150 max_angle_inclination. But then the person already knows how to fly, I want to allow him to fly with max_angle_inclination = 300... I have to open my laptop, connect the FC using USB and go CLI mode and adjust this variable, losing time.

The Rate profile seams to be a nice solution. It's easy to implement? I'm a C# and PHP programmer. I have some skill on Arduino too, but never attemped to build a Cleanflight HEX. Should I try to do it for myself? Or you guys think it's too much for a newbie?

rb1205, 2016-08-29:
I'm still convinced that, following your example, it would be easier to leave max_angle_inclination to 300 and limit the channel range on the radio to 50% than doing it via profiles.

danarrib, 2016-08-30:
How do I limit channel range in FlySky FS-i6 TX? I guess my radio doesn't support this feature.

rb1205, 2016-08-30:
I'd be surprised if it doesn't, it's probably the most important setting in a programmable TX. See your manual, function "End points" (page 22 on the version I've downloaded)

closed 2017-02-25 by hydra