cleanflight/cleanflight #2368: NMEA GPS support at 4800 baud

roskoant, 2016-08-25:
Hello!
So I have a GPS receiver with NMEA support. The problem is that most of the old GPS receivers that use the NMEA protocol, work at 4800 baud rate, which is currently not supported. My board is NAZE32 rev6 and I can't see a reason why slower speeds than 9600 are not supported. OK, it'll be a little slower, I know, but it can be added. Will it be added to the firmware, or what can I do to add it manually?

roskoant, 2016-08-28:
Noone? Is that hard to add support for 4800 baud on the hardware serial port of the Naze32? It won't consume more resources, nor it will affect the performance...

wx4cb, 2016-11-24:
Its possible to add support.

A couple of places you would have to change at least is src/io/gps.c addin a line into the struct and adding one to GPSMAX_BAUD (i think it is)

I too would like to see this too

closed 2017-02-25 by hydra