cleanflight/cleanflight #2371: Flight control in PosHold issue

xdigyx, 2016-08-26:
Hi,

Not sure if devs are interested to correct possible issues for GPS modes, however while in PosHold mode I noticed a strange behaviour:
It seems that software corrects error position of the model every approx 1 sec. While flying it is clearly visible that a copter inclination in pitch/roll changes few degrees but not smoothly but only every second as it can be seen on the log print screens below:
[image]

[image]

xdigyx, 2016-08-26:
I have just experienced another issue:
My model suddenly started to fly away in the PH mode. I checked the bbox logs and although nr of sats were consistent, a distance between following readings was more than 14meters...

andre92, 2016-08-27:
If you need better gps performance, you may want to try this cleanflight fork: https://github.com/iNavFlight/inav
Cleanflight gps has little glitch detection and doesn't use the accelerometer to estimate position between gps frames. Inav has much improved on this (at the cost of slower looptimes on F1 targets)
There are plans to merge this fork back into cleanflight in the future but I don't know when exactly.
I would start with version 1.1, as version 1.2 is not released as stable yet.

xdigyx, 2016-08-27:
I have tried iNav 1.1, however found it not such stable in flight even though I used same lfp vals.
Alt hold in CF is quite fine, just need to set 0.998-0.999 for control by acc over baro (at least it works for me).
I was hoping for just a litle/fast changes (ie throw away GPS reads while distance between last and following reading is more than say 2-3min PH mode and increase refresh rate for control as it seems to be every 1sesc.

hydra, 2016-08-27:
@xdigyx Please help out with the iNav project.  When that all flies good we'll get the merge going.

xdigyx, 2016-08-27:
I have submitted some issues in iNav project (either flight performance and GPS glitch as iNav also got this problem), but for now they are busy with another more critical issue and I understand this. Unfortunately I am not able to point them out why exactly flight performance is worst or neither change the code by myself.
However it is unknown when they will be able to solve it. I believe GPS glitch problem is important for  safety reasons (model can hit a tree or building and fall down and hurt someone).
So I had a hope that you, hydra, probably can quite fast sort out this issue together with GPS glitch (as mentioned in my previous message).

hydra, 2016-09-03:
@xdigyx I'm super busy with other tasks and GPS is not my speciality.  @digitalentity care to comment on this?

There are also some fixes that have just been applied to the codebase so please retest with a new development build.

xdigyx, 2016-09-04:
thanks for reply.
I'd say It's not that much about GPS flight mode itself, more about the CF code. PH function applies the corrections, but with freq of approx 1s, so most probably the problem is with frequency of executing a part of PH that applies error correction (this could be the first thing to look at) or could be a kind of deadband (so the correction is not applied if the error is not over a specified value), or....

Latest iNav seems to work better than the one I had, so for now I am using it.

digitalentity, 2016-09-04:
@hydra. @xdigyx  the problem with CF code is that it acts on GPS only, corrections are actually calculated and applied at 5/10Hz rate (depending on GPS), what you see as ~1Hz oscillation is merely a over-regulation in position PID control. I wanted higher frequency corrections and that's the very reeason INAV fork was started.

@xdigyx development of INAV 1.2 mostly focused on general flight performance, new code greatly improve INAV flight performance (to CF/Betaflight) levels, so you might want to try out latest INAV release candidates. There are a lot of tasks related to navigation performance enhancments, they all wait until 1.2 is released. 

I wonder why you encounter GPS glitches so frequently, it's usually a very rare occurance. We'll look into better glitch detection in INAV once we get the 1.2 out - it's very close now and I'll greatly appreciate your help testing the new code and providing logs of GPS glitches.

xdigyx, 2016-09-05:
@digitalentity as you may know I already use it and truthfully speaking is ways better than 1.1.

not sure why, I use quite common Beitian BN-880. At the beginning I had original settings and later loaded some settings provided from ardupilot web (and was a bit better). So either settings could be to work on and glitch detection.
We already have such topic at iNav repo and there we will continue. I also would be grateful sorting this out (for safety reasons).

closed 2017-02-25 by hydra