cleanflight/cleanflight #2379: Slightly wrong LED strip colors

BlueEsc, 2016-08-31:
**Synopsis**: Slightly wrong LED strip colors, discovered while experimenting with single LEDs connected to WS2811 controller. A short code review revealed following minor glitches:

In `io/ledstrip.c`, function `applyLedFixedLayers`, line 507

`int hOffset = HSV_HUE_MAX;`

This line sets hue offset to 359 degrees instead of 0 (or 360). This leads to one degree of distortion, which is clearly visible if you use single LEDs with base colors R, G or B. Wrong LEDs are dimly lit.

`int hOffset = 0;`

fixes the problem and should not lead to side effects.

**Additionally** in `common/colorconversion.c`, function `hsvToRgb24`, line 41

`int base = (isat * val) / 256;`

This line reduces precision as `base` should be an evenly distributed value from 0 to 255.

`int base = (isat * val + 127) / 255;`

fixes the problem.

With both changes applied LED strip colors are correct.

ledvinap, 2016-08-31:
`int hOffset = HSV_HUE_MAX + 1;`. Positive value is necessary to handle subtraction / modulo correctly

hydra, 2016-09-03:
thanks for bug report and solution guys.  fix merged.

commit 0fa6619c (2016-09-03): Fix led strip incorrect colors. Closes #2379. Fix by @ledvinap.
closed 2016-09-03 by hydra