cleanflight/cleanflight #2380: Green LED blinks erratically when FC is mounted upside down

sejnub, 2016-09-02:
FC: Flip32 10DOF

When the **FC is mounted upside down, the green LED goes on blinking after power on**. But the blinking isn't steady. See blinking video and blinking config pic! 
Also it is very hard to arm the FC (you need a lot of tries, till it works). As soon as the FC is armed everything is fine.

When I mount the FC upside up and change the orientaton in the config accordingly everything is fine.
See not-blinking video and not-blinking config pic! 

The Gyro and ACC curves are nice and flat. VBatt flag has no effect.

This buggy behaviour is reliable.

Any idea someone?

Attachments:

blinking config pic: 
https://drive.google.com/open?id=0B3f_WdRCKIHWeWVqMjN0ZDJJd0U

blinking video: 
https://drive.google.com/open?id=0B3f_WdRCKIHWZ09ON3dtUlJEVG8

not-blinking config pic:
https://drive.google.com/open?id=0B3f_WdRCKIHWLUNLcnNqaVU1UDA

not-blinking video: 
https://drive.google.com/open?id=0B3f_WdRCKIHWNXEtRUpoc2ZGMlU

in both cases model is upright in config: 
https://drive.google.com/open?id=0B3f_WdRCKIHWWm4wdzdIdFBmS00

Here the complete config for the blinking case: 
https://drive.google.com/open?id=0B3f_WdRCKIHWSnByQkxOWUhFMEk

hydra, 2016-09-03:
provide a config dump please.

sejnub, 2016-09-03:
I thought that that is what my last link points to. Do you mean somthing else?

sejnub, 2016-09-07:
Is my attached file not the config dump? I will gladly provide any additional information I can.

roskoant, 2016-09-15:
I can confirm this! Also, when upside down it won't arm (not sure if it's safety feature). Another issue that I think is linked to this is that if in some way(external factors like wind), FC happens to be upside down in the air in **angle** mode, the quad will disarm. The last needs more testing and logging, but it's **expensive**, so I'm not going to do this. Quad fell off the sky once because this. :D

hydra, 2016-10-30:
@sejnub sorry yes, i missed that.  not had time to investigate this yet.

wx4cb, 2016-11-24:
I use a naze32 rev 6 upside down with my roll trim set at -180.

I have also noticed erratic led and arming with the sticks. I now use a switch on my dx6 to arm so i can get voice prompts from the tx.

closed 2017-02-25 by hydra