cleanflight/cleanflight #2382: More precision for scaleRange function

BlueEsc, 2016-09-05:
**Synopsis**: Code analysis reveals that frequently used function `scaleRange` lacks a bit of precision. Differences are neglible if `destMin` to `destMax` covers a large range. But they grow in small ranges and especially affect max and min values (`destMax` is only returned if `x` is exactly `srcMax`).

In `common/math.c`, function `scaleRange` is defined in lines 168ff as follows:

```
int scaleRange(int x, int srcMin, int srcMax, int destMin, int destMax)
{
    long int a = ((long int) destMax - (long int) destMin) * ((long int) x - (long int) srcMin);
    long int b = (long int) srcMax - (long int) srcMin;
    return ((a / b) - (destMax - destMin)) + destMax;
}
```

Because of integer division truncation results are not evenly distributed between `destMin` and `destMax`. Following change of the last line leads to better results:

```
    return ((a + (b >> (long int) 1)) / b) + destMin;
```

Results are now evenly distributed in the range from `destMin` to `destMax` and therefore more precise. I checked all occurences of `scaleRange` and saw no code contradicting this change.

But please note that also the tolerance of the function is changed and evenly distributed if `x` should exceed its boundaries.

ledvinap, 2016-09-05:
IMO desired operation depends on context, current implementation is mathematically cleaner and less surprising (rounding down is implicit behavior in C).

One problem with rounding implementation is that it does not preserve strict upper bound - value ` x < srcMax` may result in `destMax` (bad for array indexing)

Also rounding won't work for negative values

BlueEsc, 2016-09-05:
It does work with negative values, which are often used for scaling.

And in my opinion it is better if an `x < srcMax` leads to `destMax` if it is appropriate. If you have a small array which you want to index according to throttle or simular you would expect indices to be evenly distributed instead of the last index only used at absolute maximum. Also applies to LED brightness or simular usages.

ledvinap, 2016-09-05:
Division in C discards fractional part (rounding towards zero). Adding half the divisor won't work then
`(-4 + 1) / 2 == -1`

In C common indexing is zero based, 0..(N-1). Applying scale_range on index will produce valid index again - `srcMax` is not in input interval, so destMax is never returned.

Mapping `0..(srcMax-1)` to `0..(destMax-1)` is probably working fine. (`scaleRange(x, 0, 256, 0, 360); ` to map byte value (0..255) to hue (0..359))

BlueEsc, 2016-09-05:
> Mapping 0..(srcMax-1) to 0..(destMax-1) is probably working fine. (scaleRange(x, 0, 256, 0, 360); to map byte value (0..255) to hue (0..359))

I agree. But unfortunately function `scaleRange` was not used with this insight. It was understood and is used as if `scaleRange(x, 0, 255, 0, 359)` would lead to correct results (according to your example).

ledvinap, 2016-09-05:
IMO it's better to fix the use.

This PR will only mask the problem by shifting output value by 0.5 - `0` and `destMax` 'buckets' will contain only half of input values in comparison with other numbers (which may make sense in some situations).
Correct use of this function will result even distribution

hydra, 2016-10-30:
@ledvinap I could not find that scaleRange index usage in the led strip code.

hydra, 2016-10-30:
@BlueEsc could you create some unit tests and a PR for this please?  I'm all for better precision and clarification of expected usage scenarios.

ledvinap, 2016-10-30:
@hydra: at least create second version of scale-range - one for `x1..(x2-1) -> y1..(y2-1)` and one for `x1..x2 -> y1..y2`

Also negative value offset/division bias should be fixed first

closed 2017-02-25 by hydra