cleanflight/cleanflight #2386: CF freezes when I save + reboot

orange-rc-pilot, 2016-09-06:
I'm using Naze32 rev 5 funfly with 1.13.0 stable firmware. Every time I hit the "save and reboot" button, it saves, reboots and then just stays like that. I click on another tab and it loads infinitely, and then the only thing I can do is disconnect the board. When I reconnect, all my settings are saved luckily.

hydra, 2016-09-07:
does it do this every time?

please provide a config dump.

do you have another board you can also verify this on?  perhaps you have a bad board?  we have not had any other reports of this.

orange-rc-pilot, 2016-09-07:
Yes, it does do it every time. I'm at work now, but when I get home I'll give you the dump. My board was working before... Perhaps something went wrong. But on the rc groups forum, there are many other people with the same problem. I posted a thread there too, to see if there are any answers. 

Thanks for your help!

hydra, 2016-09-07:
has it always done it or just started happening?  if it's just started happeneing it's possible that something in chrome changed, google, apple and microsoft have a habit of breaking stuff.  might be firmware but if it suddenly started happening then check your computer before the firmware.

orange-rc-pilot, 2016-09-07:
It only just started. I have updates set to automatic, so I don’t know when chrome last updated. I’m using a mac, but I have bootcamp installed on it so I’ll try it on Windows too.

Here’s the dump you asked for:

# version
# Cleanflight/NAZE 1.13.0 Jun  6 2016 / 00:03:48 (a77bc76)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature INFLIGHT_ACC_CAL
feature FAILSAFE
feature RX_PARALLEL_PWM

# map
map TAER1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
[... 300 lines cut ...]
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 20
set thr_expo = 0
set roll_rate = 90
set pitch_rate = 90
set yaw_rate = 95
set tpa_rate = 50
set tpa_breakpoint = 1300

Thanks again!!!

hydra, 2016-10-20:
was there a specific setting you were changing that causes the problem?  or with that config (above) does it always do it?

orange-rc-pilot, 2016-10-20:
I'm not too sure. I was fiddling around with the PIDs, doing regular tuning, and the problem magically fixed itself. Perhaps a specific configuration of PIDs upset CleanFlight, or maybe it was the board itself. I don't know. Either way, it works now. 

Thanks!!

Sevan.

hydra, 2016-11-17:
ok, closing.

closed 2016-11-17 by hydra