cleanflight/cleanflight #2387: Modes switch not working

labmaster, 2016-09-07:
I have a problem using the "Modes" config for arming.

1. verified that i could arm and fly with standard arming procedure (throttle low / right), works
2. checked in the receiver menu that a channel is propely assigend to an aux function (e.g. AUX1, but even tried different auxes coming in from the transmitter switches)
3. entered "Modes" configuration and selected the right AUX1 function at the ARM row. (first row)
Verified, that i see the marker jumping around when flipping the switch on the TX.
Set the green range (min-to max) to a save area where the small green marker stays when i want to have armed.

Here the problem begins:
a. Usually when the marker hits the chossen area the ARM field in the first column turns green.
This does not happen.
b. as soon as i have this configuration done, saved and restared can not longer arm the copter.
It does not more arm using the stick methor nor does it arm by flipping the switch.

Any Ideas ?

best regards,
Walter

hydra, 2016-09-07:
if it didn't work there'd be a lot of unhappy users :smile: myself included.   likely you're just setting it up wrong or something else is preventing you from arming.  please use rcgroups.com for support and once issues are confirmed raise them here.

don't forget to include config dump in future too, without it we can't check your configuration.

closed 2016-09-07 by hydra

labmaster, 2016-09-07:
> if it didn't work there'd be a lot of unhappy users 😄 myself included. likely you're just setting it up wrong or something else is preventing you from arming. please use rcgroups.com for support and once issues are confirmed raise them here.

If you would have read my post exactly you would have noticed that there is nothing that i could do wrong there.

But others have already confirmed a bug with the arminv function togehzer with using serial rx with lot if channels comming in.

Tried it with a standard ppm receicer and it works but not with the serial sbus rx.

it shows all receiver channels properly in the rx menu