cleanflight/cleanflight #2394: Virtual current sensor offset

fape, 2016-09-10:
I try to calibrate virtual current sensor. I follow the [documentation](https://github.com/cleanflight/cleanflight/blob/master/docs/Battery.md) I measured Imin=0,3A and Imax 71,5A. If I set current_meter_offset = 30 (Imin*100) then I got battery current 3A, I try with the sample data from documentation, Imin=2,8A, so I set current_meter_offset = 280, I got 28A as "disarmed" battery current. What am I do wrong?
I use CL 1.13.0 SRF3 with [spracingf3 deluxe clone](http://www.banggood.com/Upgrade-NAZE32-F3-Flight-Controller-Acro-6-DOF-Deluxe-10-DOF-for-Multirotor-Racing-p-1010232.html) 
[image]

fredmcc, 2016-09-11:
This is my setup for a 400mA offset.
Disable the legacy MultiWii Support.
And make sure to set the sesnor type to virtual with the following CLI command.
set current_meter_type = VIRTUAL

[image]

fape, 2016-09-11:
Thanks you are right. If I disable legacy MultiWii Support I got the proper number. If I enable it I got 10 times bigger. Is this the expected behavior/operation?

closed 2016-09-11 by fape