cleanflight/cleanflight #2395: Ensure new/changed PG are reset when firmware is upgraded.

hydra, 2016-09-11:
Currently new PG's are uninitialised when the firmware is upgraded.  The reset templates/functions should be used.

hydra, 2016-09-11:
@martinbudden do you have time for this?

martinbudden, 2016-09-12:
@hydra , I'll have a look at this in the next day or two. @ledvinap your eyes on this would also be good.

martinbudden, 2016-09-12:
@hydra , is this true of all PGs, or just the array types?

See https://github.com/cleanflight/cleanflight/blob/master/src/main/config/parameter_group.h#L170

ledvinap, 2016-09-12:
@martinbudden: only template array initialization is not implemented, init function must be used instead. 

Problem  is that reset is called only explicitly, but it should be called on new PGs after upgrade

hydra, 2016-09-12:
@martinbudden What caused me to notice this was the addition of the new voltageMeterConfig and currentMeterConfig items in #2396.  They were arrays.  I think also the new pilotConfig (in common/pilot.h) also showed the issue.  It's quite easy to test, just erase a flash, put latest stable version on, boot it. then upgrade to latest master and you'll see the issue if you put breakpoints on any of the new PG reset functions / templates added since 1.13.0

@ledvinap is correct, the lack of template array initialization isn't the issue here.

hydra, 2016-09-12:
We also need to verify that PG's with a new version are reset if there is a version mismatch.  We can open a new issue for that or tackle that as part of this.  e.g.  batteryConfig in #2396 is now at version 1 (instead of version 0)

hydra, 2016-09-12:
@martinbudden hi.  I'm working on configurator changes tonight.  If you have time to look at this I would *really* appreciate it as I'm snowed under right now and need all the help I can get.

martinbudden, 2016-09-12:
@hydra , I can't spend any time on it today, but I'll try and look at it tomorrow.

hydra, 2016-09-15:
Done via #2401

closed 2016-09-15 by hydra