cleanflight/cleanflight #2398: Change all internal voltage values to milliVolt

hydra, 2016-09-12:
Generally, we want the code to use units that people are familiar with, the legacy 8-bit code has external pressures in it which may have been the reason why real units are not used.  In the case of the voltage cleanflight mainly uses decivolts.  This issue is for discussion and PR's to move it to milliVolts. (mV).

@ledvinap said:

```

Cleanflight member
ledvinap added a line comment 3 hours ago

only use is

https://github.com/cleanflight/cleanflight/blob/62646021f2bc27f8404ba9ac620f5eb0f8f75a89/src/main/sensors/voltage.c#L70 :

return ((((uint32_t)src * config->vbatscale * 33 + (0xFFF * 5)) / (0xFFF * config->vbatresdivval)) / config->vbatresdivmultiplier);

33 is 10 * ADC reference voltage, 0xFFF * 5 was used to round correctly (does not make sense in current code)

IMO it is reasonable to specify scale as input divider ratio, probably in mV/V - 1000 for no divider, max value would be ~ 65V/V, 214V input range. Resolution will be better that 1/1000 (it makes little sense to use amplifier as input).

Whole equation will reduce to

src / <input range> * 3.3V * vbatscale / 100  // result in 0.1V

With optimization and correct rounding (3,3V moved to denominator, accuracy reduction is negligible, floating point is eliminated in compiller). Also 1lsb is Vref+/4096, not 0xfff as in current code

((int)src * vbatscale + (int)(0x1000 * 100 / 3.3 / 2)) / (int)(0x1000 * 100 / 3.3);

Also I would prefer return value in mV (removing * 100 in denominator)

Only problem will be that users need to calculate voltage divisor - https://en.wikipedia.org/wiki/Voltage_divider
Result is in physical units, so it should be much easier to document/calculate
```

@proggod said:

```
You guys might want to check my formula in the raceflight rc12 code, you are losing precision going from the floats and ints, and you are also not really taking into account the vdivider ratios, etc as well, since they are never exactly what you think they are and should be measured.
```

I said:

```
@proggod yes, i am aware of that. I specifically simplifed the unit testing of that code so that we can migrate to it at a later date.
```

hydra, 2016-09-12:
@ledvinap said:

```
@proggod : Can you please line relevant code? Is it this one? https://github.com/rs2k/raceflight/blob/master/src/main/sensors/battery.c#L66

Resolution of equation above is more than adequate, denominator is ~ 100000. Can't see any advantage to using float.
vbatscale resolution is ~ 3mv at max voltage, independent of voltage divider value. 1/1000 should be adequate considering other error sources.
```

proggod, 2016-09-13:
It was an extreme hack to keep them with in a 8 bit formula, however the casting of floats is required for accuracy.  I suggest you move the variables to floats, to make the hack not required.  i converted the scale from 0-255 to meaningful floats, and while this works, its just too hard to follow and read.   The most import things I have found is my formula to calculate voltage, this was done with a lot of research and trial and error.  I use the min voltage as a scale, again dirty hack, but i just needed to make it work properly in an hurry, and the vbat code was a horrible mess that  needed a total rewrite.  When you do a math in c like the following (variable * 3.3 / 1000) it can choose to convert the 3.3 to an integer before doing the math, if you don't tell it to use variable as a float.  Casting is important because not all compilers and all versions of compilers follow logical paths for doing the calculations.   casting any non float as a float, is important, and telling it to use final result as a float, even before converting it back to an integer makes a difference as well.  I can tell you without doing all the casting, the results were completely wrong.   I may have went over kill with casting, but this is a scenario where too much is more better then too little.

proggod, 2016-09-13:
uint16_t batteryAdcToVoltage(uint16_t src)  // in perfect world vbatscale should be 33 for 3.3v
{                                           //  vbatmincellvoltage is calculated by ((HIGH_RESISTOR+LOW_RESISTOR)/LOW_RESISTOR) stored as * 2
    return ((uint16_t)((float)src * (float)((float)(batteryConfig->vbatscale/3)/0xFFF) * (float)(batteryConfig->vbatmincellvoltage/2)));
// used for debugging    return(batteryConfig->vbatmincellvoltage);
}

This is the proper formula to calculate voltage.

https://github.com/rs2k/raceflight/blob/9a494a2ed263969ae3f8987cf312b1d2d02bb8fe/src/main/sensors/battery.c

ledvinap, 2016-09-13:
@proggod : 3.3V was not min battery voltage but ADC reference.
```
return ((uint16_t)((float)src * (float)((float)(batteryConfig->vbatscale/3)/0xFFF) * (float)(batteryConfig->vbatmincellvoltage/2)));
```
both `(batteryConfig->vbatscale/3)` and `(batteryConfig->vbatmincellvoltage/2)` are integer divisions, integer result is than casted to float.
The divisor should be 0x1000, not 0xfff (see STM datasheet), but the difference is negligible.

Cast to integer result is effectively `floor()`, discarding fractional part. Rounding will improve accuracy (which is quite important with 100mV resolution)

Voltage on input pin is `(1.0/4096.0) * ADC_result * VRef`.  
Battery voltage is input pin voltage * resistor divider  (`R2 / (R1 + R2)` when R1 is connected to battery and R2 to ground).
`(1.0/4096.0) * ADC_result * 3.3 * R2 / (R1 + R2)` is battery voltage (in volts), multiply with 10 and round to get correct value in 100mV unit. `R2 / (R1 + R2)` is  vbatscale, it is stored multiplied by 10 in MW code (in 0.1 units).

Code some time back: 
https://github.com/cleanflight/cleanflight/blob/1ecbdf3daeef4b73941a0e0fb05bc20ab909c902/src/main/sensors/battery.c#L51

@projectkk2glider converted it to use integer calculations and correct rounding (no precision lost)
 
@EvilBadger replaced `10` from `3.3 -> 33` transformation with two other parameters and this nonsense got merged ...

proggod, 2016-09-13:
I think we are essentially saying the same thing, I don't even know what the state of the current code is, and i used 33 instead of 3.3 because it was being stored in a char.   The biggest issue with cleanflight code is ratio is not being used at all.  So unless you have a perfect vdivider with built in ratio (Which no one does) the voltage was only right at one exact voltage.  I suggest you test the casting, I don't assume anything when it comes to C code, because different compilers and platforms always have different results.

martinbudden, 2016-09-13:
What's wrong with:
```
uint16_t batteryAdcToVoltage(uint16_t src)
{
    uint32_t ret = src;
    src *= 33 * R2;
    src /= 10 * (R1 + R2) * 4096;
    return (uint16_t)ret;
}
```
No floats required, retains maximum precision. Replace R1 and R2 by whatever the appropriate variables are.

proggod, 2016-09-13:
Honestly its too late to even think about the math, and its been a long time since I have looked at this.  if the results are the same as my formula, great.  Do it however you find is best, in Raceflight we have moved to 100% floats or doubles, so we wouldn't do it the same way any more.  You don't have the luxury of doing this, since your support F1 targets as well.  I just know I did a lot of testing before, and my formula was the only one found to be accurate.  I tested against very accurate equipment to ensure all results were accurate.

ledvinap, 2016-09-13:
@proggod : You should consider reading some introductory book on C programming. C language is standardized and current compilers do pretty good job following the standard. 
Data type promotion rules are quite simple and I can't imagine any compiler breaking them. 
Can you provide some example where there is difference between compilers?

In CF you were expected to set `vbatscale` to `(R2 / (R1 + R2)) * 10` and adjust it slightly to match actual resistor and Vcc (used as VRef for ADC) values.
The granularity is a bit too coarse (~1% with default divider), that's why I am proposing to use 1000 as scaling value (unit will be mV/V)

@martinbudden : 

It is easier to aggregate R1 and R2 into single value - `vbatscale`. 
Resistor are 1% at best and linear regulator ~ +-1% (LT1117, Chinese may be worse), so you need to calibrate the value anyway to get better accuracy. 

With 12 bit `src` and ~5 bits from `33`, only 15 bits are left for  `vbatscale` to avoid 32bit overflow (`src * 33 * vbatscale / (4096 * 10[from 3.3] * 10[in 100mV])`). Moving the 3.3 into divisor will fix this limitation and allow full 16bit (uint16_t) `vbatscale` range, with negligible reduction of accuracy (divisor is 12412 instead of 12412.12, 10ppm error)

You also want to round the result correctly - result is in 100mV units, so rounding will reduce error from 0.1V to 0.05V - quite important with few LiPo cells

proggod, 2016-09-13:
Holy shit man, i have been doing C coding for 25 years seriously it comments like that, the chase people away from helping with these projects like the plague.   Who do you think runs RaceFlight right now?

ledvinap, 2016-09-13:
@proggod : I'am very sorry. 
I got confused by https://github.com/rs2k/raceflight/blob/9a494a2ed263969ae3f8987cf312b1d2d02bb8fe/src/main/sensors/battery.c#L66, I mistook you as author of the code.

hydra, 2016-09-13:
@proggod I think @ledvinap was just trying to be helpful and probably wasn't aware of your background.  The thing with C is that you *can* be doing it for 25 years and still not know it inside out. :smile: 

I bet however that we can simply just write some unit tests that give expected adc readings and test the output and easily change the underlying formula for one which is efficient, clear and gives us the results we desire.

The first step in doing this is to update the unit tests, then anyone with an interest can go try different formulas and inspect the generated code and calculate CPU cycles used to their heart's content! :smile: 

@martinbudden @proggod @ledvinap there is a test case here:

https://github.com/cleanflight/cleanflight/blob/master/src/test/unit/voltage_unittest.cc#L53-L68

I imagine it would be trivial for any of you to update that test case and update the formula in a PR 

The build for the PR doesn't have to be fully working - just enough code + test so we can see/discuss the changes.

proggod, 2016-09-13:
Casting is never a bad thing, over-casting is a lot better then under-casting.  You are telling the compiler exactly how you expect it to work, and it will optimize any excess calls out of the code.  Not every language treats everything the same, so getting in the habit of being much more explicit in your code, is good practice.

martinbudden, 2016-09-13:
@ledvinap , I think you missed the point I was trying to make with my code sample. It was not concerned with the specifics of the vbat conversion, but more to illustrate that you could avoid casting with uptyping.

Also note code it read more often than written, so having casts tells the reader your intent.

Also note that someone with a handle of @proggod is not likely to be a novice programmer.

@proggod , personally I prefer uptyping to casting, since I think it is more readable. But I do agree with your point about making your intent explicit. I sometimes write and run code on an Arduino before I port it to Cleanflight/betaflight/iNav and I have certainly been caught out by differences around the handing of 8-bit values. I've had code that worked perfectly well on an Arduino, but did not work in Cleanflight.

proggod, 2016-09-13:
@martinbudden i work with a lot of new coders, and teaching them to write out exactly what they are thinking, always has the best results.  When working with 8bit values, C does tend to do some weird things with longer formulas, so using multiple casts, seems to tell the compiler to do exactly what you want it to do, and your right all that casting probably came from arduino at first, because its where the battery formula was originally tested.

I hope to recruit a good team to bring in some new talent into cleanflight, but we need to watch being too critical of people.  So far a lot of this issues have others talking down to programmers, and it scares them away.   I don't think its intentional, but its something to try and work hard to prevent.   Our RaceFlight following has grown huge, and I think that is due a large part to our community, and us not allowing such things to chase new people away.  We all have different opinions, and even have our own stupid moments, and no one likes being criticized when these happen.   Sometimes people may be doing things, because of experience or for reasons that are not immediately evident.

ledvinap, 2016-09-13:
@martinbudden:  
Code before 3879b6c5662f41f134360417487b7999d4ddb7b9 was: ` return ((uint32_t)src * batteryConfig->vbatscale * 33 + (0xFFF * 5)) / (0xFFF * 10); `, almost identical to your code ...

ledvinap, 2016-09-13:
The problem with https://github.com/rs2k/raceflight/blob/9a494a2ed263969ae3f8987cf312b1d2d02bb8fe/src/main/sensors/battery.c#L66 and 929e3e883b2a92dfa6694e1d01ca6e2bc296991e is not explicit casting or floating point use (but floating point is unnecessary, as shown by @martinbudden).

1. `batteryConfig->vbatmincellvoltage` is replaced by hardcoded `VBATMINCELLVOLTAGE`

2. `VRef` is replaced by `batteryConfig->vbatscale/3`. Default `vbatscale` is 110 (except on ALIENWIIF3, ALIENFLIGHTF4), so new value is 36. 20% error (unless you run FC at 3.6V) and makes no sense. Multiplier from volts to return value in 100mV is factored into this variable.

3. `vbatmincellvoltage` is documented as `((HIGH_RESISTOR+LOW_RESISTOR)/LOW_RESISTOR) * 2` (`vbatscale` is used in original code for this),  but it is defaulted to 20 (using hack in 6f0fbd911c95dcea75e63ef7f77cb1d4016ba737, ignoring different defaults on COLIBRI_RACE). With 10k/1k voltage divisor, this should be 22. At least this reduces error caused in previous term by `vbatscale/3`. 

4. both `vbatscale` and 'vbatmincellvoltage' are divided first (as integers), so resulting value is truncated and only then is cast to `float`. Absolutely unnecessary loss of precision .. 

5. Resulting value is truncated, not rounded, increasing error.

This code was raped and then repeatedly beaten until it returned acceptable value, totally ignoring logic behind it. I hope CleanFlight is not going in this direction ...

proggod, 2016-09-13:
resistors are only accurate to 5% usually, so hard coding anything is ridiculous, especially since your using to resistors in a circuit which makes the calculations completely in accurate.

I have no desire to argue about old code, or how things are done or why.  Whatever code you guys decide to do, make sure you test the results at many different voltages, with may different boards, and voltage divider circuits.

Floats are faster and more accurate then integer math on f3 and f4.

But this is a useless debate, I could care less how you decide to handle this, as long as results are accurate.

martinbudden, 2016-09-13:
@proggod , I agree we should be trying to bring new developers on board, and part of that is not being critical of people. I always try and be constructive and encouraging in my comments.

While it's important not to criticise people personally, it is important to be able to criticise code and for people not to take that personally. Everyone can write bad code on a bad day.

I also think this is a bit of a "what colour shall we paint the bikeshed?" debate. This bit of code is not really that important - there are many other pressing issues to deal with.

proggod, 2016-09-13:
@martinbudden these "discussions" like with @ledvinap are exactly what I don't need in my day, and either do others, discussing endlessly over how to do a simple line of code is as useless as watching paint dry.

martinbudden, 2016-09-13:
To be fair, @ledvinap is a very smart guy, and his criticism of my code has improved it on many occasions.

BTW are you mainly working on Raceflight, or are you planning to work on Cleanflight as well?

proggod, 2016-09-13:
@martinbudden cleanflight is falling behind because of the process involved.  There is a happy medium between doing things properly, and discuss things to death.  I have no doubt @ledvinap is a good programmer, but there are more then one way to skin a cat.   My concern is cleanflight's market share is dwindling, so some things need to change if we hope to make it relevant again.   RaceFlights market share i growing out of control, and while I believe we are not doing everything right.  We need to find a compromise between doing things clean and proper, and still moving quick enough to keep the relevant features implemented at decent speed.   We also need to remember programmers have many different code styles, and having them communicate and help each other in a less public manor makes them less likely to want to quit a project.  Its the reason we rely on slack so much, and a reason we keep developer channels private.   We don't want people to be too afraid to be seen as a fool.

@rs2k got laughed at, and everyone told him his ideas were wrong.  Until I came along he was too afraid to do things the way he felt they should be done.   People consider RaceFlight the best flying software available, and a lot of this has to do with that we aren't afraid to try things, and argue about theory later.

To answer your last question, yes myself an Dominic have been talking quite a bit on combining efforts, as soon as you release your current release we plan to push forward bringing a lot of innovation into cleanflight, and I plan to spend a lot of my time marketing and pushing cleanflight to the next level.  I really want us to work together to benefit both projects greatly.  I don't like the way people are treated on some of these projects (even if people mean no offense by it) and I don't like the way betaflight is going, and leaving cleanflight behind.   We will be helping add a lot of innovation to cleanflight, and I really hope you guys are on board with helping push it in that direction as well.

I do think a good first step is building a much better communication channel then github issues :)

martinbudden, 2016-09-13:
I think one of the reasons Cleanflight is falling behind is the @hydra has had a lot on his plate for the last month or two. But it looks like he is back in the driving seat now.

I work on Cleanflight, betaflight and iNav and will continue to work on all three projects.

I like Cleanflight because of it's "clean code" goal, and because it has working test code. My natural style of development is to write test code along with the code. I have worked with some very very good programmers in my time. Way way better than I am, and I have observed that the very best programmers also write the best test code - I try and emulate that as far as I can.

I like betaflight because it is more free to experiment. If you look at the releases you'll see that @borisbstyle has tried ideas out and thrown them away if they don't work. That's a good way to make progress.

I like iNav because I'm interested in sensor fusion and also autonomous flight. I'm also interested in using interesting sensors like optical flow.

One of the things I am working on is to converge the projects in areas where they are common, in particular the device driver, sensor and IO code - I've talked to @hydra about this before. One of the things needed to make that happen is for Cleanflight to convert to the new IO. That would enable substantial catchup with betaflight and I think should be a priority for the next release.

@hydra one of the things I think you need to do as the Cleanflight BDFL is to keep on top of the issues and PRs. There have been some good PRs that have atrophied, for example @joshuabardwell wrote a documentation PR for PID tuning.

proggod, 2016-09-13:
@martinbudden i am helping Dominic clear a lot from his plate soon so he can spend more time coding, however I think cleanflight needs a lot more then just dominic working on the code :)

martinbudden, 2016-09-13:
@proggod , I didn't say that the problem was that dominic was not spending enough time on the code.

borisbstyle, 2016-09-13:
@ledvinap 
Don't bother watching that code they are closed source project. The story goes that they are so good that they rewrote everything from scratch in three weeks.

@proggod Since when would betaflight turn back to cleanflight or any open source :D? @martinbudden is working hard to keep things in sync between good things frm iNav, betaflight and cleanflight all the time as I personally don't have much time for it. He is doing a great job.
It's all about sharing and open source ;). The more you give the more you get.

I understand about raceflights big commercial plans, but I don't see that fit here.

skaman82, 2016-09-13:
@martinbudden Well I think it is partly is. The main reason cleanflight is struggling, because it is moving way to slowly. If a person puts a big amount of free time to write a PR for CF and that PR is ignored forever, this person sure does not try again, ever and at best diverts to betaflight community. Just look at the open PRs. And if nobody writes new stuff anymore for CF, there is no movement at all and the popularity goes down. It can't be done well and fast enough by a single person. I really hope that the situation improves somehow, because I really love the CF project.

borisbstyle, 2016-09-13:
@skaman82 
Yes thats exactly how betaflight was started. To give those cluttered PR's a chance to be tried out. Just got out of the hand.

martinbudden, 2016-09-13:
@skaman82 I don't want to say too much without @hydra being part of the conversation, but what I mean was that I didn't think the reason that Cleanflight is falling behind is because Dominic is not spending enough time coding. It's as you say, it's because not enough time is being spent on the issues and PRs. As I mentioned before, there have been some good PRs that have atrophied.

Running a software project is a difficult job, but one of the things you absolutely must do is keep on top of the issue list. And you need to spend a significant amount of your time on the PRs.

skaman82, 2016-09-14:
@martinbudden agreed. I really don't want to step on @hydra s toes, I really like the work he is doing. That is just the reality as I see it. I know it is not pleasant to read criticism but we all are grownups who can handle it without hard feelings. In the end this all being sad with the intention to get CF on top again - and thats a good thing.

hydra, 2016-09-14:
Ok people, let's not get too carried away with discussions unrelated to this issue.  focus please.  I have no hard feelings, I love all the cool stuff that's come about due to the project I started.   Due to the volume of PR's and code quality of them it's not been possible to keep on top of them all.  What do you do when a big PR for a feature that someone has spent ages on just isn't up to scratch and you haven't got time to work on it because you're dealing with 50 other PR's in the same situation?

If we had a paid team of developers and a product owner I'm sure things could be different but we don't and that's the way it goes.

I have to stand by the principles that made me start cleanflight in the first place and I compromise and sometimes take shortcuts when possible but if it's not up to scratch and there isn't time to look at it then it doesn't get merged until we have time.  Without more trusted maintainers and people with similar code-quality goals to help out code quality and long term maintainability WILL suffer, it's not acceptable to me to just merge code because someone has spent time writing it and people want it - that is a recipe for disaster.

Right now this issue is low priority.  The IO changes are more important to me, even though some people (@proggod, @rs2k) have reservations with @ledvinap solution there are still massive benefits to code sharing between the projects now that the iNav and Betaflight have adopted them.

Anyway, let's get back on track - voltages!

hydra, 2016-09-14:
@martinbudden if there are some documentation PR's that should be merged then now is the time since we're releasing v1.14.0-RC2 soon.  Please add any PR's to issue #2211

mlyle, 2016-09-22:
@proggod Whut?

You're incorrect on the facts of how type promotion (not demotion) work in C.  You should expect to be called on it and not go off on a tirade when it's shown you're wrong.

> Holy shit man, i have been doing C coding for 25 years seriously it comments like that, the chase people away from helping with these projects like the plague. Who do you think runs RaceFlight right now?

http://raceflight.net/team/

That is super impressive.  I don't know many 18 year olds that have been programming in C for 25 years.  You can confidently say you have more experience than anyone else your age.

hydra, 2017-03-10:
re-opening this as it has a lot of merit still

closed 2023-04-19 by hydra