cleanflight/cleanflight #2399: OSD improvements

hydra, 2016-09-12:
Now that the first release of CleanflightOSD is on shipping production boards there is likely to be feedback from users, this issue is a placeholder so we can gather feedback from users on what they like, dislike and want in the future.

I already know that they want:
* Ability to configure PIDs.
* Ability to flash/show a message or highlight remaining battery capacity when it is running out.

They might want:
* Ability to configure rates.
* Symbols / Graphics to instead of letters (e.g. for RSSI, AMP, MAH, BAT, 12V, 5V) to conserve screen space.
* Symbols for flight modes.
* Ability to easily switch battery capacity - I regularly fly with a set of 1300 and 1500 packs on the same quad.
* Artificial horizion.
* Heading/Compass display.
* Altitude bars.
* Ability to dedicate an area for their own logo.  e.g. 3x3 grid
* Flight summary.

There are many things that we *could* add to an OSD but I want the users to specifically tell me what they want so I don't waste time adding things that people don't care much about.

krzysztofmatula, 2016-09-14:
I'd vote for:
- artificial horizon
- altitude display + graphical vario
- home distance and home direction indicator (GPS)

I can't wait to see it in action. Having a fast dedicated processor should allow for a really smooth graphics. What will be the supported screen refresh rate btw? Is 50Hz achievable?

hydra, 2016-09-14:
Possible yes, there is some work to be done to allow 50hz - the main thing being the timing of the display updates to the vsync signal.  Let's keep this topic focused on feature additions and not get bogged down with other discussions.

jonasbae, 2016-09-18:
- Support for Side by Side 3D video - Like the signal comming from the blackbird2 for example

LitterBugs, 2016-10-17:
I like most everything that Hydra said we "Might want".  May be some omissions or duplication below and some unique ideas for future development:
Switch  passthrough from FC to OSD for selecting different layouts or select configuration menu when not armed
Integration with intelligent video TX for band, channel, frequency, and power changing  (ie TBS Unify Pro smart audio control)
Small or Custom field labels and symbols.  Currently takes too much space.
Ability to change Font size for each field on the screen.  (highlighting the fields I look at most)
Customization of flight summary.
Ability to left, center, or right justify fields and labels
Center cross-hair even when artificial horizon is off.
Baro and/or GPS altitude
Configure/change battery capacity from configuration menu
Set points for warnings on low average cell voltage, Amp draw, mah drawn
Countdown Timer started at Arm.
Configure/Change LED colors from configuration menu
Auto cell count detection and display
Display Watts
Graphical and/or numeric Vario
GPS coordinates
GPS speed 
FC CPU utilization
Throttle %
Direction to Home arrow and distance
Graphical and/or numeric heading via Compass and/or GPS
Change opacity of OSD (if it's not a hardware limitation)
Lap counter and timer?  Would need reverse transponder input of some sort.

Just a few for now.  

Cheers!
LitterBug

SandroGrzicic, 2016-10-23:
Hi,

here's my list of top 10 features (most of them already mentioned above):

1. GPS speed
2. Symbols / Graphics to instead of letters (e.g. for RSSI, AMP, MAH, BAT, 12V, 5V) to conserve screen space.
3. FC CPU utilization
4. Font size adjustment, ideally per-element; a set of 2 or 3 options would be perfect (e.g. small/medium/large)
5. At powerup, you can move your roll/pitch stick left/right/top/bottom to select one of up to 4 battery profiles (like rate profiles), e.g. 1300 vs 1500, and then each profile has an associated mAh and maximum flight time which can count down to zero (profiles are of course configured in advance via the configurator).
6. Baro and/or GPS altitude
7. Set points for warnings on low average cell voltage, Amp draw, mah drawn
8. Flight summary.
9. Artificial horizon
10. GPS coordinates (primarily useful as a lost model finder backup)

Thanks for the great work so far on the OSD!

neildotwilliams, 2016-11-19:
I think litterbugs eluded to this. I would like to see EVERYTHING I can while unarmed (moon phase, probability of et existence, rotational speed of Mars,etc). And only a few select things while armed (rssi and voltage). Possibly change osd objects via transmitter switch OR have at least 2 profiles: armed and unarmed.

neildotwilliams, 2016-11-19:
Flash the whole screen at preset points. Eg rssi low. Battery near critical. Or flash a big warning in the middle.

unitware, 2017-01-20:
I'd like:
+1 for artificial horizon and altitude bars, with configurable position to match camera tilt and lens
- vtx power to read in mW
- possibility to filter available vtx bands/channels on the toggle band/channel button function.
- possibility to toggle band/channel from the transmitter

closed 2017-02-25 by hydra